mriscoc/Ender3V2S1 #799: [BUG] Dont' disable motors

fabiomaticus, 2023-03-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

After updating the firmware to the latest 20230312 version, at the end of a print the motors remain blocked.
I have in gcode M84

### Version of Professional Firmware

2.1.3 20230312

### Printer model

Ender 3S1

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

badaszszsz, 2023-03-17:
@fabiomaticus what if you try to send `M18`?

fabiomaticus, 2023-03-18:
I have 3 ender 3 S1 printers. The problem was present on all. After restarting them, the problem is solved

closed 2023-03-18 by fabiomaticus