mriscoc/Ender3V2S1 #802: Print head bed crash with execution of G28 in GCode

BradGillies, 2023-03-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When a print is completed and the machine is left in that state and a second print is started without doing a Homing from the screen.  the print head will crash the bed due to the Z probe not extending (CR TOUCH)
Here are the first lines of the Printer Start GCode from Cura.

; Ender 3 Custom Start G-code
G92 E0 ; Reset Extruder
G28 ; Home all axes
G29 L0; Bed Level

As I am quite new to this, it could well be I have something incorrect in the setup or firmware that is causing the issue.  The stock Creality firmware did not exhibit this behaviour.
The printer is an Ender 3 V2 with a Sprite Pro Extruder

If I run the Homing process from the screen the homing works as expected.  It only seems to occur when the homing is initiates from the Gcode.

### Bug Timeline

This has been ther since loading the firmware

### Expected behavior

Expected the homing process wor work with the CR Touch operable

### Actual behavior

CR Touch probe does not deploy causing a bed crash

### Steps to Reproduce

1. complete a print 
2. immediately start another print with no other intervention 
3. during the homing part of the code the z probe does not deploy and a bed crash results

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 V2 Modified

### Electronics

4.2.2 board

### Add-ons

Second Z-Axis Stepper, Sprite Pro Extruder

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

badaszszsz, 2023-03-17:
check if your gantry is grounded (it caused mine and some other people problems) https://github.com/mriscoc/Ender3V2S1/discussions/413#discussioncomment-5278026

Slyv06, 2023-03-27:
Hello, I have tested this firmware two days ago on ender 3 v2 and a new "sprite extruder pro". Sometimes all goes well... sometimes the probe does not deploy (-> beautifull crash bed ..) and even sometime it doesn't try to level on Z axis ( it starts to print as if the homing and leveling was already done)
Even forcing the probe deployment into the gcode, the steps are not runned...

mriscoc, 2023-03-29:
Check connections, disable HS mode.

BradGillies, 2023-03-29:
HS Mode is confirmed off and connections confirmed.

 

Rolling back to stock firmware corrects the issue.

badaszszsz, 2023-04-14:
Do you have ferrites on cables? Do this happen every time or just rarerly? Do you have washers added to CR Touch and correct mount of CR Touch relative to nozzle tip?

doctorevil30564, 2023-04-17:
I am having the same basic issue, unless HS is enabled by default in the configuration for the source code, it isn't enabled on mine either. I know I didn't specifically look for this setting to turn it on.

I didn't see this bug, or I wouldn't have posted in the general discussion to see if anyone else was having this same problem.

doctorevil30564, 2023-04-17:
I am having the same basic issue, unless HS is enabled by default in the configuration for the source code, it isn't enabled on mine either. If I power off the printer, then turn it back on it works for the first print after powering it back on.

I didn't see this bug issue, or I wouldn't have posted in the general discussion to see if anyone else was having this same problem.

doctorevil30564, 2023-04-17:
I am having the same basic issue, unless HS is enabled by default in the configuration for the source code, it isn't enabled on mine either. If I power off the printer then turn it back on it works for the first print after powering it back on.

I didn't see this bug, or I wouldn't have posted in the general discussion to see if anyone else was having this same problem.

I will check my cable by swapping it out for the other new one I have and make sure all connections are snug and secure.

mriscoc, 2023-04-19:
> HS is enabled by default in the configuration

Leave HS mode enabled in the configuration file, otherwise it might break some library assumptions.

closed 2023-06-28 by github-actions[bot]