mriscoc/Ender3V2S1 #806: [BUG] unknown syntax error for send gcodes via COM earlier received by M503 (v20230312)

badaszszsz, 2023-03-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

C125 X0 Y230 Z40 was read from M503 via COM. There are other's not working too.

Cannot restore settings via COM sending the command to firmware, previously backuped with M503

### Bug Timeline

n/a

### Expected behavior

Command received from M503 works to load into RAM and are saved with M500

### Actual behavior

Not works. In comparison to M413 S1

>>> M413 S1 ; ON
SENDING:M413 S1 ; ON
ok P15 B15
>>> C125 X0 Y230 Z40
*** Unknown syntax: C125 X0 Y230 Z40
>>> C412 M0 ; Active mode: LOW
*** Unknown syntax: C412 M0 ; Active mode: LOW
>>> C562 E0 ; No inverted
*** Unknown syntax: C562 E0 ; No inverted
>>> C104 T275
*** Unknown syntax: C104 T275
>>> C29 L25.00 R201.50 F25.00 B205.00 N9 T85 ; 9x9T=85°C
*** Unknown syntax: C29 L25.00 R201.50 F25.00 B205.00 N9 T85 ; 9x9T=85°C
>>> C102 X235 Y235
*** Unknown syntax: C102 X235 Y235
>>> C100 X-2 Y-2
*** Unknown syntax: C100 X-2 Y-2
>>> C101 X243 Y230 Z250
*** Unknown syntax: C101 X243 Y230 Z250
>>> C851 S960 M5
*** Unknown syntax: C851 S960 M5

### Steps to Reproduce

connect via USB to COM
get Pronterface and send `M503` code
try to send back all the gcodes after setting defaults back (`M502`?)
send commands back (additionally, you could read back and try to store&read again and compare i.e. in Meld diff app)

### Version of Professional Firmware

Ender3V2-4322-BLTUBL-MPC 2.1.3 20230312

### Printer model

Creality Ender 3V2 Neo

### Electronics

4.2.2

### Add-ons

no hw modifications

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

Pronterface

### Additional information & file uploads

[image]

badaszszsz, 2023-03-17:
or probably needs this...

https://github.com/mriscoc/Ender3V2S1/wiki/Configuration-files

mriscoc, 2023-03-17:
Please read the note in the wiki about Pronterface and non-standard G-codes. It is an Pronterface issue, not a firmware issue.

closed 2023-03-17 by mriscoc

badaszszsz, 2023-03-18:
https://github.com/mriscoc/Ender3V2S1/wiki/Custom-C-gcodes

Yup, right.

mriscoc, 2023-05-17:
There is now a special Pronterface version with support for the Professional firmware custom G-codes:
https://github.com/mriscoc/Pronterface-PFEd/releases/tag/PFEd-20230517