mriscoc/Ender3V2S1 #819: screen freezes when accessing media, displays media list, printer reboots

spectangulation, 2023-03-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

the printer randomly reset, on 2 occasions across approx. 10 prints when accessing media.

### Bug Timeline

new bug? it has happened twice. i do not know if the conditions were the same

### Expected behavior

i expected my media to refresh and to start a print

### Actual behavior

instead, the printer froze, reset itself, and shut off the heat, but i was able to start my print once it booted up again.

### Steps to Reproduce

1. preheat on
2. media update on
3. access media
4. see no items (list was previously long)
5. scroll wheel up, see items
6. screen freezes
7. printer reboots
8. screen still frozen on media list
9. firmware toast pops on top of media list
10. screen unfreezes

no settings changed, 

### Version of Professional Firmware

latest

### Printer model

ender 3v2

### Electronics

422

### Add-ons

n/a

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-03-25:
Keep low the number of files on your memory card. Format your memory card, change it for a better quality/speed one.

spectangulation, 2023-03-26:
> Keep low the number of files on your memory card. 
is there a general limit i should maintain? I dont delete anything off my SD unless they are test files

RiverSharks, 2023-03-26:
I have similar issue on my Ender 3 S1 `[latest F4 firmware](https://github.com/mriscoc/Ender3V2S1/releases/download/20230312/Ender3S1-F4-20230312.bin)`, it occasionally freezes and reboots as author explained, I haven't found correlation with media menu. It rebooted in Advanced menu once and in Prepare.

mriscoc, 2023-03-27:
> is there a general limit i should maintain?

I didn't test for the limit, but each listed file takes memory and there is not too much memory in the control board of a 3D printer, but I suggest no more of 50 files in the SD card.

TheWebMachine, 2023-03-27:
For what it is worth, I have not had any reboots like this. I only use the SD card for laser (octoprint for FDM), but I've got a lot of files on the SD card. The only differences I can think of are the files are relatively small and don't contain preview images. Not sure if any of that matters. I could load up a ton of gco's on to an SD and do some testing to see if I can find a pattern. I have a multi-day print running, so I'll give it a go in a couple days.

spectangulation, 2023-03-27:
> ...I suggest no more of 50 files in the SD card.

I have 18 files on my SD, total

>  The only differences I can think of are the files are relatively small and don't contain preview images.

I don't have preview images. the largest file is 14mb (25pc print), and 8 of the other 17 files are between 1 and 8mb; the rest are under 500kb.

I suspect the issue lies in Media Update being on, or worst case, the SD extension i have.

TheWebMachine, 2023-03-27:
Good to know. I'll test with media update on and off. 

> the SD extension i have.

SD extender or SD adapter (like Micro to full-size)? Is it a cable-to-slot or card-to-card?

spectangulation, 2023-03-27:
> SD extender or SD adapter (like Micro to full-size)? Is it a cable-to-slot or card-to-card?

it's a microSD to SD extension, and i'm using a microSD in an SD adapter:
https://www.amazon.ca/dp/B07JMTW1YD?psc=1&ref=ppx_yo2ov_dt_b_product_details

TheWebMachine, 2023-03-27:
So it's going SD card -> SD-to-Micro extender -> Micro-to-SD adapter -> original SD slot on the printer case?

TheWebMachine, 2023-03-27:
This is reminiscent of issues I've had with I2C in a project recently. I suspect you may be encountering data errors. As you add connectors and cable length to the data line, you add capacitance and resistance to the wiring, which affect the rise/fall of binary signals. If these timings stretch too much, data will become corrupt in transit.

The difference you are seeing in reliability between firmwares could be down to max clock speeds on the SPI bus set in the firmware. (SD cards use SPI, which is another communication tech similar to I2C.) A firmware with a lower SPI clock set would have higher tolerances with cable length/capacitance/resistance. Higher SPI clock would be far more sensitive to this.

spectangulation, 2023-03-28:
> So it's going SD card -> SD-to-Micro extender -> Micro-to-SD adapter -> original SD slot on the printer case?

It's going: Printer microSD port -> SD Extension -> microSD in SD adapter

this is a common mod to ender machines, especially those whose machines are located on the left side of their workspace.

TheWebMachine, 2023-03-28:
> this is a common mod to ender machines,

Agreed. I just wonder if maybe the SPI clock is at a higher default speed in this firmware compared to others, which could be a factor with such a long extension and multiple adapters/connections enroute. It only came to mind because this firmware already has a higher supported Serial port baudrate that doesn't play well with some cheap long USB cables. 

Of course, I could also be completely wrong here. 🤷🏻‍♂️😁

spectangulation, 2023-03-28:
i did notice this runs at 250k baud, though it does have a slower option. i am not knowledgeable enough on either the hardware or software to understand the difference, or if there is any impact to this issue.

it doesn't happen often, and it hasn't happened since I posted this.

TheWebMachine, 2023-03-28:
Fair enough. I'm sure @mriscoc will set me straight on the SPI train of thought when he gets a chance. SPI clock and the Serial port baudrate are totally separate, but I don't know if he opted to increase SPI clock (for faster SDcard reads) when he increased Serial baud for this build.

mriscoc, 2023-03-28:
> higher supported Serial port baudrate that doesn't play well with some cheap long USB cables.

Just to remind, USB connection has its own protocol and its speed doesn't depends on the RS232 baud rate, the programmed serial speed (250000 baud) is between the USB chip and the SoC in the control board. That said, increasing the baud rate increase the number of data packets in the USB protocol but not the speed of protocol itself.

mriscoc, 2023-03-28:
The communication speed between the memory card and the board has not changed from the speed specified in Marlin.

TheWebMachine, 2023-03-28:
> > higher supported Serial port baudrate that doesn't play well with some cheap long USB cables.
> 
> Just to remind, USB connection has its own protocol and its speed doesn't depends on the RS232 baud rate, the programmed serial speed (250000 baud) is between the USB chip and the SoC inside of the printer control board. That said, increasing the baud rate increase the number of data packets in the USB protocol but not the speed of protocol itself.

Indeed...sometimes, in my attempt to curb my long posts, I gloss over the finer details. haha In the super cheap long usb cables and under "normal" usage, errored packets between a PC and device would just get retransmitted unbeknownst to the end user...but the more "real-time" nature of the printer's serial needs during a print has been known to cause issues (too many resends, stalls in print, etc) with octoprint, especially at higher baud. 

The moral of any such story being: Use the shortest and highest quality cable and connectors that you can, especially when random failures are prone to being costly on their own (wasted filament in this case) and/or less tolerated by the equipment or protocol (latency sensitive devices, serial, I2C, SPI, etc).

(And in the case of I2C, _never_ forget your pull-up resistors! 🤦🏻‍♂️🤣)

darth-crunchus, 2023-04-05:
> > is there a general limit i should maintain?
> 
> I didn't test for the limit, but each listed file takes memory and there is not too much memory in the control board of a 3D printer, but I suggest no more of 50 files in the SD card.

This happens even with far fewer than 50 files on the card.  As an example, I have 22 gcode files across 5 directories (Love that it supports organization, btw) (Calibration, Calibration\Tramming, Maintenance, Normal Prints, Printer Parts), and it still does it if I try to access the media after completing a print.  It may have something to do with having an extender cable, I don't know, but that seems like the firmware is ejecting the media?  I don't even know the firmware is capable of performing a software-level media eject.

spectangulation, 2023-04-05:
good to see im not the only person this has happened with. Are you also using the Media Update function?

darth-crunchus, 2023-04-06:
Not sure, if I am, actually, I know I was on the most recent version prior to the current version, but I don't know I have it turned on in this fw.  I'll let you know in about 6.5 hours or so; she printin'.

spectangulation, 2023-06-10:
honestly it still happens but idc, really. it's w/e.

classicrocker883, 2023-07-23:
> honestly it still happens but idc, really. it's w/e.

hey youre still having this issue? i think i seemed to have solved it.  
in **/src/lcd/e3v2/common/encoder.h** look for `#define ENCODER_WAIT_MS` and change the value to `10`.

I noticed since the recent update, it would cause the screen to freeze and restart after clicking the encoder too fast in the media menu when looking for a file to print. I think Marlin has this set to `20` and at some point it was `10`, then changed back to `20`.

change this to `10` and see if that works. if you cant compile your own check out [my repo release here](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-2-ender3)

closed 2023-09-11 by mriscoc