mriscoc/Ender3V2S1 #822: [BUG] the cooler fan cannot work automatically

leezoneline, 2023-03-26:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

the cooler fan cannot work when i set the cooling options in Cura. Whatever the value i set in the fan speed option, it just cannot work automantically.

### Bug Timeline

every time when i start to print

### Expected behavior

the cooling fan can work

### Actual behavior

it didn't

### Steps to Reproduce

1.Slice a model
2.Enable cooling options
3.It will happen

### Version of Professional Firmware

2.1.3-20230312 with LA

### Printer model

Creality Ender 3 S1F4

### Electronics

STM32F401RC

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

Cura

### Additional information & file uploads

[settings.zip](https://github.com/mriscoc/Ender3V2S1/files/11071265/settings.zip)
sorry i don't know how to make log output, plz explain that to me.

mriscoc, 2023-03-26:
Verify sending the G-code [M106](https://marlinfw.org/docs/gcode/M106.html) using a console as Octoprint terminal or Pronterface.

leezoneline, 2023-03-27:
It works at 255. Should I do anything else?

leezoneline, 2023-03-28:
But I still failed to set the fan speed in Cura, It just don't work when starting printing.

mriscoc, 2023-03-28:
If the printer works with the M106 you could have a problem in your slicer/settings not in the firmware. There isn't reports about the part fan issues with [Experimental-Ender3S1-F4-LA-20230312.bin](https://github.com/mriscoc/Special_Configurations/releases/download/LinearAdvance/Experimental-Ender3S1-F4-LA-20230312.bin)

This Issue Queue is for Firmware bug reports and development-related issues, and we prefer not to handle general printer questions here. For best results getting help with configuration and troubleshooting, please use the following resources:

[Marlin Documentation](https://marlinfw.org/) hosted on MarlinFW.org
[Marlin Firmware for 3D Printers](https://www.facebook.com/groups/3Dtechtalk/) Facebook Group
[Marlin Firmware](https://discord.gg/n5NJ59y) Discord server

Also check in your printer manufacturer social media or in your slicer forums.

closed 2023-03-28 by leezoneline