mriscoc/Ender3V2S1 #843: SKR Mini 3 V3.0 compile failure with Ender3V2S1-20230312

I-wish-2, 2023-04-04:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

Using the Configurator.pyw in the Special_Configurations-main.zip Folder.  I generated this configuration, TestE3S1-427-BLTUBL-IS-LA-MPC.
It compiled successfully with both versions of the Ender3V2S1-20230312.zip and the Ender3V2S1-Ender3V2S1-Release.zip firmware.

Again using the Configurator.pyw in the Special_Configurations-main.zip colder.  I generated this configuration, TestE3S1-SKRME3V3-BLTUBL-IS-LA-MPC.
Both versions of Ender3V2S1-20230312.zip and the Ender3V2S1-Ender3V2S1-Release.zip releases Failed to compile with error message shown below.  This compile failure is very repeatable.

### Bug Timeline

New for me, today 4/4/2023

### Expected behavior

A successful compile using Visual Studio Code, Ender3V2S1-20230312.zip or Ender3V2S1-Ender3V2S1-Release.zip, and the Configurator.pyw program.

### Actual behavior

The compile will repeatable fail with these the following error messages:

from Marlin\src\gcode\feature\input_shaping\M593.cpp:28:

Marlin\src\gcode\feature\input_shaping\../../../module/stepper/../../feature/../lcd/e3v2/proui/../../../module/thermistor/thermistors.h:24:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory
   24 | #include "../../inc/MarlinConfig.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.

*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\feature\input_shaping\M593.cpp.o] Error 1

### Steps to Reproduce

1. Use these current files, Special_Configurations-main.zip, Ender3V2S1-Ender3V2S1-Release.zip or Ender3V2S1-20230312.zip.
2. Using the Configurator.pyw in the Special_Configurations-main.zip colder.  I generated this configuration, TestE3S1-SKRME3V3-BLTUBL-IS-LA-MPC.
3. Compile with Visual Studio Code, the compile fails with the above noted errors.

### Version of Professional Firmware

Version 2.1.3 20230403  8:25

### Printer model

Creality Ender 3 S1

### Electronics

SKR Mini 3 V3.0

### Add-ons

Ender 3 V2 Dwin Display

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

### Additional information & file uploads

There are no additional edits to any of the firmware.

mriscoc, 2023-04-05:
What option in the Configurator.pyw doesn't work?

I-wish-2, 2023-04-05:
Greetings,  I do not have any Configuratior.pyw option issues as it work great.  The problem is with the compile failing.  I'm using the latest version of the Special configurations and source code 20230403 as displayed on the printer info screen.

Here is the problem with the compile.
The Configuration E3V2-427-BLTUBL-T13-IS-LA-MPC-NCDF  This configuration compiles just fine with no errors.
The Configuration E3S1-SKRME3V3-BLTUBL-T13-IS-LA-MPC-NCDF  Fails during the compile with this error message.

"Marlin\src\gcode\feature\input_shaping\../../../module/stepper/../../feature/../lcd/e3v2/proui/../../../module/thermistor/thermistors.h:24:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory"

Only with the SKRME3V3 motherboard selection.
D

mriscoc, 2023-04-05:
I just test the generated configuration files and it compiles without issues.

[image]

Compare these configuration files with yours:
[E3S1-SKRME3V3-BLTUBL-T13-IS-LA-MPC-NCDF.zip](https://github.com/mriscoc/Ender3V2S1/files/11159751/E3S1-SKRME3V3-BLTUBL-T13-IS-LA-MPC-NCDF.zip)

This is the generated firmware:
[E3S1-SKRME3V3-BLTUBL-T13-IS-LA-MPC-NCDF-20230405.zip](https://github.com/mriscoc/Ender3V2S1/files/11159763/E3S1-SKRME3V3-BLTUBL-T13-IS-LA-MPC-NCDF-20230405.zip)

I-wish-2, 2023-04-05:
Hi, I loaded your current configuration files in the current source code and had the same error when I compiled it.

I have flushed all of my current loaded source and configuration unzipped files and am starting again from scratch.  I must have something corrupted.  Results to follow:  Thanks Again for your help and I have saved your configuration and bin files to test with.
: )
D

I-wish-2, 2023-04-05:
Ok here  we go,
I did a clean install of both the source and special configuration zip files.  I did a complete reinstall of Visual Studio Code ver 1.77.1.
I used the configurator to build my SKRME3V3 configuration and the compile failed.
I used your SKRME3V3 configuration and it failed as well.  The NCDF.json file works great in both cases.
In addition I installed your SKRME3V3 bin file in the E3S1 and it works great.  I'm saved!!

I then used the configurator to build an E3V2 427 with all the trimmings and it compiled great. Go Figure.

The error from the failed SKRME3V3 compiles is always the same:

"Marlin\src\gcode\feature\input_shaping\../../../module/stepper/../../feature/../lcd/e3v2/proui/../../../module/thermistor/thermistors.h:24:10: fatal error: ../../inc/MarlinConfig.h: No such file or directory"

*** [.pio\build\STM32G0B1RE_btt\src\src\gcode\feature\input_shaping\M593.cpp.o] Error 1

Miguel at this point I'm going to run your bin file and have made a backup up copy.  This is to crazy for me to figure out and as you have given me a current firmware running printer, Thank You Again, I'm going to toss in the towel for the moment and maybe an answer will pop up some time later.  My brain is tired.
Thanks Again for all the help and if you think of something else I'm all ears and ready to test it.
As Always My Best Regards
Doug

mriscoc, 2023-04-05:
Try to shorter the path of the Marlin installation, without spaces of special characters. For the SKRME3V3, Marlin needs to download some libraries and save in the disk.

I-wish-2, 2023-04-06:
Ok will do, is that the spaces in the configurator file name?.  Just one other thought.. are we using the same compiler application and OS ?  I'm using Win10 and Visual Studio Code.

I-wish-2, 2023-04-06:
I tried the compile with just this name E3S1SKRME3V3.  It failed as well although I might not be understanding what you are requesting with " Try shortening the path of the Marlin installation, without spaces of special characters. "
D

I-wish-2, 2023-04-06:
Is it possible for me to download the libraries that Marlin needs and save then to the correct folder?
D

I-wish-2, 2023-04-06:
I did try this configuration Ender3S1-SKRME3V3-BLTUBL-T13 with and without the dashes and both of the compiles failed as well with the same error message as before.
D

SorenRR, 2023-04-06:
>I am curious though in how to install a program OUTSIDE of the "Programs/Program Files" location. Is that >done when the install asks where to install the program? Thanks again

Yep... Very few programs do not ask. Back in the day we used 8.3 name notation, that's when "C:\Program Files" would be referenced to as "C:\progra~1". This started with Unicode names/directories in Windows 95 and Windows NT. "dir /x" will list 8.3 names next to long names if more than 8 characters or special characters is used ;-)

mriscoc, 2023-04-07:
Where is your Marlin Project folder? Check if your path is something similar to: C:\Ender\Marlin

I-wish-2, 2023-04-07:
I understand now and am correcting mine as I was 7 folders down from the C: drive.  It is now C:\Firmware\Use Me E3V2S1-20230312 and being tested.  Thank You.
D

I-wish-2, 2023-04-07:
BINGO we have success with the compile.  OMG, Miguel your a genius and really on top of this firmware and its requirements.  Thank you for staying the course with me to resolve this issue.  You always go above and beyond to help.
Thank you so much.  As always my Best Regards
Doug

I-wish-2, 2023-04-07:
Let's close this issue as it was not a firmware issue just a file management folder one.  Thanks again D

closed 2023-04-07 by I-wish-2