mriscoc/Ender3V2S1 #847: [BUG] building mesh with 8x8 or 9x9 fails at the 64th point

wbeck77, 2023-04-08:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hi all,

i am running a Ender 3 S1 with the newest professional firmware. Generally works fine, thank you!

Doing the bed leveling mesh with e.g. 5x5 or 7x7 points also works properly.

But if I try to do so with 8x8 or 9x9 it fails doing the 64th point. 
It's always the 64th point. So 63 are running smoothly.

Maybe a kind of SW issue here? 64 its a power of 2 so could be a bug?

Best regards

Wolfgang Beck 

### Version of Professional Firmware

20230312

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-04-08:
The answer is in the Wiki, please always look for similar questions.
Duplicate of: #594, #485, #285, #203, #85, #32

closed 2023-04-08 by mriscoc

wbeck77, 2023-04-08:
Many thanks for your quick support! 
And sorry, I simply had not the idea that someone else would have faced this issue in the past already.

I disabled HS mode and now it wents through even in 8x8 and 9x9

One question. if the Bltouch is the issue would it be worth to change the default config to "disabled HS mode"?
At least for Ender 3S1. I am using your firmware precompiled and "out of the box" and at least for me it was enabled.

mriscoc, 2023-04-08:
> One question. if the Bltouch is the issue would it be worth to change the default config to "disabled HS mode"?

Not many people make a mesh larger than 5x5 because it is enough for a 230x230 bed size. In that case, HS can be kept on with the benefit of a faster mesh creation.