mriscoc/Ender3V2S1 #848: Bed tramming, with mag mounted, probe keeps stowing probe at each point

Fieldy1999, 2023-04-08:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When using the bed traming wizard, with a magnetically mounted probe, the probe is stowed and deployed at each probe point. I.e The printer deploys probe, probes point, stows, returns to centre then deploys probe again and moves to next probe point.

### Expected behavior

Have the printer keep probe and only stow once all points have been probed.

### Steps to Reproduce

Execute bed tramming with a mag mounted probve

### Version of Professional Firmware

20230312

### Printer model

Ender 3 V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-04-08:
The deploy/stow mechanism in the bed tramming wizard was thought to prevent collisions with the tip of the CR/BLTouch. For other probes really it doesn't have sense.

Maybe a flag could be implemented in the future for avoid stow between probe points.

TheWebMachine, 2023-04-22:
> The deploy/stow mechanism in the bed tramming wizard was thought to prevent collisions with the tip of the CR/BLTouch. For other probes really it doesn't have sense.
> 
> Maybe a flag could be implemented in the future for avoid stow between probe points.

In all honesty, the moves the tramming wizard makes are actually quite slow and inefficient. It moves to position, raises Z, deploys probe, touches, raises Z again, stows probe, lowers Z, **then** moves to next corner, raises Z **again**, deploys probe...rinse, lather, repeat. It makes **three** Z moves per corner! 

It makes more sense to skip the extra lower Z before moving to the next corner which would prevent needing to stow the probe *and* be faster by eliminating TWO of the Z moves per corner, as subsequent corners won't need a preceding raise before deploying/probing, whether you are stowing the probe or not.

Perhaps tie the flag for tram probe stowage to the HS Probe menu option and clean up the tramming wizard to be more efficient by eliminating excessive Z moves?

mriscoc, 2023-04-23:
The extra Z raise is from the the Z clearance for deploy. There is a issue if Z is not raised when the user change from full manual to automatic, I have a nice line in the bed as a reminder of that.

TheWebMachine, 2023-04-23:
Hmm...i think I get what you mean. There is no difference between choosing a single probe tran corner and the full wizard. So, changing the Z raise for full Wizard would affect individual probed corners...which, in turn, could cause damage in the full manual to probe transition you mention.

Well, at the very least, perhaps link the tram stow behavior to HS Mode toggle?

mriscoc, 2023-04-23:
> Well, at the very least, perhaps link the tram stow behavior to HS Mode toggle?

Certainly, it is possible to enhanced the behaviour and optimize the movements in the bed tramming wizard, but, that will add code and not represents any printing improvement, it is only a few seconds delay. So, is it worth spending program memory fixing that?

TheWebMachine, 2023-04-23:
> Certainly, it is possible to enhanced the behaviour and optimize the movements in the bed tramming wizard, but, that will add code and not represents any printing improvement, it is only a few [...]

Certainly not on the F4! 🤣🤣🤣😞😭

mriscoc, 2023-04-23:
I will try to do something that have minimum impact in the program memory usage.

TheWebMachine, 2023-04-23:
Personally, it doesn't bother me (I ABL way more often than I tram), but if there is a way to simply link the tram stow to the HS mode toggle, it would satisfy the OP's request.

mriscoc, 2023-04-24:
I think that it is fixed now in my private testing environment with a cost of 40 bytes of additional program memory.

TheWebMachine, 2023-04-24:
Sounds like that was way too easy for you. 😜  Now, can we get these printers to play Crysis? I bet you could get it in there in like 60 bytes...

closed 2023-07-04 by github-actions[bot]