mriscoc/Ender3V2S1 #852: [BUG] Valid mesh point not probed if CR Touch outside bed *before* travel to point

badaszszsz, 2023-04-13:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

[BUG] Valid mesh point not probed if CR Touch outside bed *before* travel to point 
I define valid mesh point defined as coordinates of nozzle inside the bed area.
I use CR Touch for Ender 3 v2 Neo with CR touch mounted with stock holder on the left side of hotend caret.

If the point is near X0 or Y0 axis of the bed, because of negative value of Probe offset, Software check if the probe is inside the bed boundaries. But before the translation to current nozzle position.

That disallow automatic bed mesh measurement to probe far places of the bed. This result with possibility to use only part of bed (as printing near boundaries would be risky, cause offset increase there much).

### Bug Timeline

I seen it in last 2 releases, not check others; 20230312 SKRE3v3 beta IS and precompiled & 422 BLTUBL precompiled ; 20221222 422 BLTUBL precompiled

### Expected behavior

Mesh level attempt to probe valid point after travel of probe (as others points) in scope (whole N x N points coordinates). Possibility of Z auto probe with Edit Mesh 

### Actual behavior

No attempt in auto leveling of probe. When attempting to Edit Mesh to auto probe one by one, points described above result in no probe and error message "Z Probe Past Bed"

### Steps to Reproduce

1. Maximize mesh area with Advanced / Mesh Leveling / Mesh Inset with requirement the mesh edge points relative nozzle-probe offset make probe be outside of bed
2. Start Auto level and assert some points not measured
3. save mesh
4. go to Edit Mesh, select one of not measured points
5. Probing Z Reference 

### Version of Professional Firmware

Ender3V2-SKRME3V3-BLTUBL-MPC-20230312.bin

### Printer model

Creality Ender 3V2 Neo

### Electronics

SKR Mini E3 V3

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Other (explain below)

### Host Software

Other (explain below)

### Additional information & file uploads

Your Slicer: I use Cura but report not related to slicer used
host sw: I use SD (headless) or Pronterface, but for this I only use DCIM with knob (unrelated with this in my opinion, tried with SD card in/out and without Pronterface nor USB cable connected)

Thanks for your hard and awesome work, @mriscoc! 🎉 Happy to be here

badaszszsz, 2023-04-13:
...and `M115`

```
>>> M115
SENDING:M115
FIRMWARE_NAME:Marlin 2.1.3 MRiscoC Ender3V2-SKRME3V3-BLTUBL-IS-MPC, based on bugfix-2.1.x (Mar 12 2023 19:33:12) SOURCE_CODE_URL:github.com/mriscoc/Ender3V2S1 PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender3V2-SKRME3V3-BLTUBL-IS-MPC EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:1
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_POS:0
Cap:AUTOREPORT_TEMP:1
Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:RUNOUT:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:1
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:1
Cap:HOST_ACTION_COMMANDS:1
Cap:PROMPT_SUPPORT:1
Cap:SDCARD:1
Cap:MULTI_VOLUME:0
Cap:REPEAT:0
Cap:SD_WRITE:1
Cap:AUTOREPORT_SD_STATUS:0
Cap:LONG_FILENAME:1
Cap:LFN_WRITE:1
Cap:CUSTOM_FIRMWARE_UPLOAD:0
Cap:EXTENDED_M20:1
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:ARCS:1
Cap:BABYSTEPPING:1
Cap:CHAMBER_TEMPERATURE:0
Cap:COOLER_TEMPERATURE:0
Cap:MEATPACK:0
Cap:CONFIG_EXPORT:0
area:{full:{min:{x:-3.00,y:3.00,z:0.00},max:{x:242.00,y:232.00,z:250.00}},work:{min:{x:0.00,y:3.00,z:0.00},max:{x:235.00,y:230.00,z:250.00}}}
ok P15 B15
```

mriscoc, 2023-04-13:
> I think that yes. I did this to use those 12mm both sides

Maximize mesh takes all the available workspace, if you changed manually the mesh insets is very possible that some mesh points fall in unreachable positions. You can test that positions using the G-code [G30 Xn Yn](https://marlinfw.org/docs/gcode/G030.html)

closed 2023-06-24 by github-actions[bot]