mriscoc/Ender3V2S1 #853: #error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."

PharmCat, 2023-04-13:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Firmware not compiled

```
G:\Cloud\Software\ender3s1\Ender3V2S1-Ender3V2S1-Released>platformio run --silent -e STM32F401RE_creality
Marlin\src\lcd\e3v2\proui\dwin.cpp:175:4: error: #error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."
  175 |   #error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."
      |    ^~~~~
In file included from Marlin\src\lcd\e3v2\proui\dwin.cpp:98:
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void Draw_Tune_Menu()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:3194:69: error: 'SetPwrLossr' was not declared in this scope
 3194 |       EDIT_ITEM(ICON_Pwrlossr, MSG_OUTAGE_RECOVERY, onDrawChkbMenu, SetPwrLossr, &recovery.enabled);
      |                                                                     ^~~~~~~~~~~
Marlin\src\lcd\e3v2\proui\menus.h:53:59: note: in definition of macro 'EDIT_ITEM'
   53 | #define EDIT_ITEM(I,L,V...) EditItemAdd(I, GET_TEXT_F(L), V)
      |                                                           ^
Marlin\src\lcd\e3v2\proui\dwin.cpp:3194:83: error: 'recovery' was not declared in this scope; did you mean 'remove'?
 3194 |       EDIT_ITEM(ICON_Pwrlossr, MSG_OUTAGE_RECOVERY, onDrawChkbMenu, SetPwrLossr, &recovery.enabled);
      |                                                                                   ^~~~~~~~
Marlin\src\lcd\e3v2\proui\menus.h:53:59: note: in definition of macro 'EDIT_ITEM'
   53 | #define EDIT_ITEM(I,L,V...) EditItemAdd(I, GET_TEXT_F(L), V)
      |                                                           ^
*** [.pio\build\STM32F401RE_creality\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1
```

### Bug Timeline

Bug

### Expected behavior

Compile

### Actual behavior

Not compiled

### Steps to Reproduce

Download https://github.com/mriscoc/Ender3V2S1
Download https://github.com/mriscoc/Special_Configurations
Used Ender3S1-F4-LA
move Configuration.h and Configuration_adv.h files to the Marlin folder 
move the platform.io file to the root

### Version of Professional Firmware

Marh 13

### Printer model

S1

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

```
G:\Cloud\Software\ender3s1\Ender3V2S1-Ender3V2S1-Released>platformio run --silent -e STM32F401RE_creality
Marlin\src\lcd\e3v2\proui\dwin.cpp:175:4: error: #error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."
  175 |   #error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."
      |    ^~~~~
In file included from Marlin\src\lcd\e3v2\proui\dwin.cpp:98:
Marlin\src\lcd\e3v2\proui\dwin.cpp: In function 'void Draw_Tune_Menu()':
Marlin\src\lcd\e3v2\proui\dwin.cpp:3194:69: error: 'SetPwrLossr' was not declared in this scope
 3194 |       EDIT_ITEM(ICON_Pwrlossr, MSG_OUTAGE_RECOVERY, onDrawChkbMenu, SetPwrLossr, &recovery.enabled);
      |                                                                     ^~~~~~~~~~~
Marlin\src\lcd\e3v2\proui\menus.h:53:59: note: in definition of macro 'EDIT_ITEM'
   53 | #define EDIT_ITEM(I,L,V...) EditItemAdd(I, GET_TEXT_F(L), V)
      |                                                           ^
Marlin\src\lcd\e3v2\proui\dwin.cpp:3194:83: error: 'recovery' was not declared in this scope; did you mean 'remove'?
 3194 |       EDIT_ITEM(ICON_Pwrlossr, MSG_OUTAGE_RECOVERY, onDrawChkbMenu, SetPwrLossr, &recovery.enabled);
      |                                                                                   ^~~~~~~~
Marlin\src\lcd\e3v2\proui\menus.h:53:59: note: in definition of macro 'EDIT_ITEM'
   53 | #define EDIT_ITEM(I,L,V...) EditItemAdd(I, GET_TEXT_F(L), V)
      |                                                           ^
*** [.pio\build\STM32F401RE_creality\src\src\lcd\e3v2\proui\dwin.cpp.o] Error 1
```

PharmCat, 2023-04-13:
This may be solution [Configuration_adv.h.txt](https://github.com/mriscoc/Ender3V2S1/files/11225955/Configuration_adv.h.txt) , didn't check... but this compiled...

mriscoc, 2023-04-14:
Not a firmware error, the sources indicates that your configuration files are wrong:

#error "PLR_TUNE_ITEM requires POWER_LOSS_RECOVERY."

closed 2023-04-14 by mriscoc

mriscoc, 2023-04-17:
> may be POWER_LOSS_RECOVERY should be enabled in Ender3S1-F4-LA

No, the F4 SoC doesn't have enough program memory available.

PharmCat, 2023-04-17:
> > may be POWER_LOSS_RECOVERY should be enabled in Ender3S1-F4-LA
> 
> No, the F4 SoC doesn't have enough program memory available.

So, maybe PLR_TUNE_ITEM should be disabled in Configuration.h  ?

mriscoc, 2023-04-18:
> So, maybe PLR_TUNE_ITEM should be disabled in Configuration.h ?

Yes, it should be disabled for Ender3S1-F4-LA because POWER_LOSS_RECOVERY was disabled. I just saw that the configuration files for that version are wrong, thanks.