mriscoc/Ender3V2S1 #865: [BUG] Aquila Autobuild firmware doesn't work

brentrichardjr, 2023-04-16:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Auto Build firmware using GD32F103RC_voxelab doesn't work on the aquila.
- Doesn't restart after install.
- Manually restarting doesn't boot into firmware.

Auto Build firmware using GD32F103RC_voxelab_maple works as expected.

Pulled repo today (4/16/23). Used configurator.pyw to generate config files (Ender3S1-G32-BLTUBL-T13)

I'll try to respond to any questions the best I can. I'm new to compiling my own firmware. Thanks for all you do!

### Version of Professional Firmware

2.1.3 Mriscoc

### Printer model

Voxelab Aquila

### Electronics

GD23F103

### Add-ons

Sprite Pro Extruder; CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

mriscoc, 2023-04-20:
If you use the Configurator, the default environment configured in the Platformio.ini for the G32 version is GD32F103RC_voxelab_maple, using that environment is correct.

classicrocker883, 2023-04-25:
if youre looking for Voxelab Aquila specific

**[check out this repo here](https://github.com/classicrocker883/MriscocProUI)**

I was able to get it to work for the GD32 and N32 chips, and the interface be more like Alex's. be aware that these chips must use "_Maple_" configuration and "*RC*" `default_envs` in **platformio.ini**

the chips cannot flash with anything more than 256k, technically 89.1% of 256k but yeah. if you want `ProUIex` and *UBL* with all the added features, you will have to disable some things in the *Marlin/Config...h** files. the *_maple* env uses more memory space but there is nothing you can do since it wont flash without it. *Manual Mesh* and all the other features should fit fine. *UBL* seems to take up the most memory.

closed 2023-07-06 by github-actions[bot]