mriscoc/Ender3V2S1 #867: [BUG] (bug summary)

RigoHoward, 2023-04-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Im just overwhelmed on how many useful options and how its slmost everything perfect on this firmware, but i encountered and annoying issue which i will describe in the following lines...

When im printing, the extruder stepper motor do a loud noise when it's constantly retracting the filament.
I tried several retraction reduced values in the slicer (also in the firmware retraction menu too) and i couldnt get rid of the loud noise at all. But what its even worst, i founded as a consecuence with the excesive retraction movements, the filament starts to whorn out and the gears finally doesnt grab them anymore, resulting in no filament extruded from the nozzle after a few layers.

I try using jyers firmware and also the official creality 1.0.7 firmware and i dont get the noise or the filament whorn issue at all so its very odd.

### Bug Timeline

I noticed the issue since the first use of the mriscoc firmware

### Expected behavior

When the extruder motor its doing retraction moves you hear a slight sound from the gears and everything works fine

### Actual behavior

The extruder stepper motor do a loud noise everytime it retracts the filament

### Steps to Reproduce

1. Slice a stl file with latest version of Cura (let the retraction settings with the default values)
2. Put the gcode on your microsd
3. Start your print on your ender 3v2 with the latest release of mriscoc firmware
4.  Hear the noise of the extruder stepper motor when it does retraction movements
5. The print fails because after a few layers no more filament its extruded by the nozzle

### Version of Professional Firmware

20230312 

### Printer model

Ender 3V2

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

I really like s lot this firmware but i cant get any successful print using it

mriscoc, 2023-04-19:
For lower the noise for that particular extruder try decreasing the speed and acceleration of the extruder in the Control/Motion menu.

RigoHoward, 2023-04-20:
Oh you were right! thank you.

I take the deafult values for speed and accelaration from jyers firmware, and now im printing well without the noise on the extruder.

Here are the values if anyone is interested...

extruder max acceleration 1000
extruder max speed 25

closed 2023-04-20 by RigoHoward