mriscoc/Ender3V2S1 #871: [BUG] preview glitch

playrobotics-alex, 2023-04-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I am using Prusa slicer 2.52 with the CreateHeaderAndThumbnail.py script to generate the preview file.
Everything works but the preview image is not displayed correctly :
[image]

I don't have any other issues with the lcd (Ender s1 F4 printer) the qr code and everything else is shown correctly...
Am I doing something wrong?

### Version of Professional Firmware

Ender3S1-F4-20230312

### Printer model

S1 F4

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

playrobotics-alex, 2023-04-19:
I did open the gcode file and found the thumbnail base 64 code.
I decoded it using an online tool and the image itself looks good
[image]

mriscoc, 2023-04-19:
You could have a TJC display unit, that screen has currently some limitations. Next version of the firmware will have a workaround.

playrobotics-alex, 2023-04-19:
Yes I believe I have the TJC display.
I also noticed I don't see the numbers on the circled during leveling
Ok will be patiently waiting for the next version :-)

mriscoc, 2023-04-21:
Not a firmware issue, TJC us currently incompatible (check the wiki) with the latest release version.

closed 2023-04-21 by mriscoc

mriscoc, 2023-06-22:
TJC displays need an square preview thumbnail and it is supported in the latest firmware version.