mriscoc/Ender3V2S1 #876: [BUG] (Filament unloads at start of print; runout is disabled)

cds333, 2023-04-21:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

At the start of a print while the extruder is supposed to be doing its first purge line, the extruder cranks up in reverse and purges the entire filament. 

Enable runout is off (box is unchecked), it is set to HIGH (but that shouldn't even matter since it is disabled). All the numeric values in the filament menu are set to 0.0mm.

### Bug Timeline

Has been happening for multiple firmware versions.

### Expected behavior

Expected a purge line on the bed.

### Actual behavior

Extruder completely unloads filament. 

### Steps to Reproduce

No abnormal steps required. 

### Version of Professional Firmware

E3V2-427-BLTUBL-MPC v2.1.3 20221222

### Printer model

Ender 3 Pro

### Electronics

4.2.7

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

Beginning of gcode file that causes the malfunction- (Cura 5.0.0)

M140 S60
M105
M190 S60
M104 S200
M105
M109 S200
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
;G92 E0 ; Reset Extruder
G28 ; Home all axes
M420 S1
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
;G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:9
;LAYER:0
M107
G0 F6000 X97.163 Y96.243 Z0.2
G0 X90.313 Y108.855
;TYPE:SKIRT

mriscoc, 2023-04-21:
Try with the latest version: [Ender3V2-422-BLTUBL-MPC-20230312.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230312/Ender3V2-422-BLTUBL-MPC-20230312.bin)

ThorstenWirth, 2023-04-25:
Your slicer might work in absolute positioning mode.
M82 and M83 have solved this problem for me.
I use this start g-code:

;; ########## Ender 3 Custom Start G-code ########## ;;
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M105
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
G28 ; Home all axes
;; Load bed mesh from eeprom
;M420 L0 S1 Z2 ; Loading the bed mesh
;; End loading mesh
;; Create mesh at start
G29 P1 ; Do automated probing of the bed.
G29 P3 T ; Repeat until all mesh points are filled in.
;G29 S1 ; Save UBL mesh points to EEPROM.
G29 F 2.0 ; Set Fade Height for correction at 2.0 mm.
G29 A ; Activate the UBL System.
;M500 ; Save current setup. WARNING: UBL will be active at power up, before any [`G28`](/docs/gcode/G028.html).
;; End creating mesh
M104 S{material_print_temperature_layer_0} ; Set Extruder temperature
M105
M109 S{material_print_temperature_layer_0} ; Wait for Extruder temperature
M83 ; switch to relative extrusion
G1 E-2 F1500
G1 E2 F1500
G1 E22 F2400 ; Load material
G1 X0.1 Y.1 Z3.0 F5000.0 ; move to off the edge
G1 X0.1 Y20 Z3.0 F5000.0 ; move to start-line position
G1 Z3.0 F3000 ; move z up little to prevent scratching of surface
G1 Z0.3 F3000 ; move z back down a bit
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
M82 ; switch back to absolute extrusion
G92 E0 ; reset extruder
G1 Z2.0 F3000 ; move z up little to prevent scratching of surface
C108 ; Close the mesh viewer (optional)
;; ########## End of custom start GCode ########## ;;

cds333, 2023-04-27:
Is there a known cause that was patched? 

I know there were similar issues related to filament runout, but since mine is off it cant be that. 

The fact that the exact same file will cause the malfunction one print and then not the next would lead me to think the slicer is not to blame. 

Thanks for all your hard work. I absolutely love everything else about this firmware and install it whenever I can.

mriscoc, 2023-04-28:
> Is there a known cause that was patched?

Each version is based in the latest Marlin bugfix available at compile date, so maybe for that release there was some change that fix your issue. But there are no reports about problems with the runout sensor other than HIGH/LOW misconfiguration.

closed 2023-07-08 by github-actions[bot]