mriscoc/Ender3V2S1 #879: [BUG] (bug summary) No longer connecting to Pronterface

KobraNeo, 2023-04-21:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I attempted to connect my Ender 3 S1 to pronterface to pull my mesh to view in a mesh viewer and it never connected. 
Updated to Professional F4 non UBL last night and made a great torture toaster and now my bed is warped beyond usability and I cannot connect to view what parts of my bed need work. 

### Bug Timeline

New bug, 4/21/23

### Expected behavior

I expected to connect on Port Com3 at Baud 115200 just like I did last night before the install

### Actual behavior

Never connected, kept recieiving "got rubbish reply from com 3"

### Steps to Reproduce

1. Install Software
2. Connect computer to 3d printer with usb
3. fail to connect

### Version of Professional Firmware

Ender3S1-F4-20230312

### Printer model

Ender 3 S1

### Electronics

If I knew I wouldnt be here

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

Crazy-Charles, 2023-04-21:
Think the port speed by default on this firmware is 250000

KobraNeo, 2023-04-21:
I wont be finding out; I was within the 14 day period and took it apart to return it. 3d printing isnt for me, I'd rather buy an injection molded version of something that wont split after waiting 14 hours for it to finish haha

mriscoc, 2023-04-21:
Not a firmware issue

closed 2023-04-21 by mriscoc