mriscoc/Ender3V2S1 #881: [BUG] Random Reboots F4-CVLM-20230407

TheWebMachine, 2023-04-22:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Multiple hours into a print, the printer randomly reboots. Doesn't appear to be a "red" error; just a crash of FW that results in reboot.

### Bug Timeline

New with F4-CVLM-20230407

### Expected behavior

Print completion

### Actual behavior

Reboot/failure

### Steps to Reproduce

1. Start Print
2. Wait many hours
3. Die a little inside when your 26 hour print fails in the 18th hour

### Version of Professional Firmware

Experimental-Ender3S1-F4-CVLM-20230407.bin

### Printer model

E3 Plus (F4)

### Electronics

F4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

This has happened a couple times now, with prints of various sizes. Octoprint is chugging along just fine for hours, then the printer randomly reboots in the middle of a standard G1 move. Octoprint merely sees serial disconnect. Printer boots without red screen error and looks like a normal boot.

The first print it happened in was only expected to be about 3 hours and it failed at about 2.

This time, it was a 26 hour print that didn't fail until after the 18th hour... which wasted a quarter KG of expensive flexible filament. 😭 I was a few feet away when it happened and all I heard was the fans ramp up to 100% and the normal boot beep. I didn't see the screen - too far away and it was in B0 off state anyway. 

I'm going to roll back to my last known working FW (20230315) and run this same 26 hour print again, but I expect it will turn out fine. I was running 20+hr prints on 20230315 without incident for weeks.

I'll post more information as I discern it, but wanted to get the ball rolling on this issue in case anyone else can add to it from their own experience.

TheWebMachine, 2023-04-22:
Final lines of Octoprint:
````
Send: N2009047 G1 E1 F1800*27
Recv: ok N2009047 P0 B15
Send: N2009048 G1 F1473.913*93
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:3997
Please see https://faq.octoprint.org/serialerror for possible reasons of this.
Changing monitoring state from "Printing" to "Offline after error"
Connection closed, closing down monitor
````

mriscoc, 2023-04-22:
It seems to be a hang, did you do another operation before printing?

TheWebMachine, 2023-04-22:
> It seems to be a hang, did you do another operation before printing? Do you have the +VUSB removed?

This is a super recent issue and *nothing* about my setup has changed. I have only two theories at the moment:

1. a new bug not on 20230315
2. My recent inclusion of M256 B0 to turn the display completely off

I'm presently testing #2. I seem to remember a long while back that when I turned the LCD off, the printer would occasionally randomly reboot, even if it wasn't printing. I'm not sure if perhaps turning the display off does more than simply turn the brightness to 0. I recently added M256 B0 to my start code after perusing a recent discussion about it...which is also right around the time I moved to the latest FW...and when this problem started happening.

Either could be the culprit at the moment, so I'm going to test both possibilties.

And, no, I've never had my +VUSB removed, but my printer (along with camera) is plugged into a fully externally powered USB3 Hub, then to Octoprint. So power fluctuations from the Pi's USB ports would have no effect on the printer...and this random reboot wasn't ever a problem outside of this new version or the recent attempt to print with LCD fully off. 

Since the LCD Off may be at play here, I'll note that I have the DACAI screen running the firmware/assets linked in this repo.

TheWebMachine, 2023-04-23:
Well, over 14h in with the same gcode file and same FW but LCD On this time and no hangs or reboots so far...only 12h and some sanity to go. 

If it survives this 26hr print, it has to be LCD Off causing the hang. I printed multiple prototypes on this firnware before I started turning the LCD Off with `M256 B0` at the start. Only after this did I notice a couple prints fail with this type of disconnect from Octoprint. However, only the latest one was I nearby to rule out things like Octoprint crashing or other external factors like jam or flex filament extruder strip/overheat. l

I think there is something about either the F4 boards or DACAI screen that evetually hangs if the LCD is set B0 during a print.

Does anything else happen in firmware beyond LCD backlight power being tuned to 0V when B0? Does it also suspend updates to the LCD itself or something like that?

TheWebMachine, 2023-04-24:
It's almost certainly LCD Off/B0 that is causing the hang. My print just finished successfully. 30.5hrs without incident (so much for the slicer's 26hr estimate lol) with the LCD on.

I'm going to do a little more testing - with shorter prints and *much* cheaper filament 😭 - to confirm. I had hangs as short as 2hrs into a 3.5hr print recently. 

I bet something else is done beyond just brightness being lowered to nothing. If the firmware is suspending comms to the display when LCD Off, it could be revealing a bug there...like board trying to send data to a suspended LCD and glitching out randomly when it doesn't get a response. 🤷🏻‍♂️

mriscoc, 2023-04-24:
`M256 B` only controls the LCD backlight and does not suspend display activities. The screen is powered by the +5V rail, so lowering the brightness only have impact in the +5V, I don't have the schematics for the F4 board, but in the 422, the +5V is connected to the +VUSB.

TheWebMachine, 2023-04-24:
Looks like I might have to pull out the oscilloscope and do some checking. I'll also butcher a USB cable to see if anything wonky is happening with that. I'll see what I come up with over the next week or so. I just really needed to get this big expensive print out of the way. Thankfully, the big failed print wasn't a total loss. 😎

tystuyfzand, 2023-05-04:
Unfortunately I can confirm this issue without any special screen blanking/etc.

Latest version of the firmware for Ender 3 v2 with 4.2.2 board + ABL (I'll get specifics a bit later) and the same message occurs randomly during prints. Sometimes after 5 minutes, sometimes after 10 hours, sometimes never.

TheWebMachine, 2023-05-04:
Well, so far, I have had no hangs/reboots if the LCD remains on. I only *randomly* get hangs/reboots on the F4 board if the LCD is off. Sometimes it takes an hour, sometimes 18 hours, sometimes the print finishes fine regardless of length...but the display is **always off** when the hang/reboot occurs.

@tystuyfzand I think you are experiencing a different issue and should open a different bug report. The 4.2.2 board is an F1 board and a little different than the F4 in my Plus. (F4 has an FPU, for example)

tystuyfzand, 2023-05-04:
> @tystuyfzand I think you are experiencing a different issue and should open a different bug report. The 4.2.2 board is an F1 board and a little different than the F4 in my Plus. (F4 has an FPU, for example)

I'm still curious if it's the same issue. It's so intermittent it's possible that the display isn't the issue (I've done full prints just fine, and had prints fail within 10 minutes), but I'll be sure to open a new issue regardless.

TheWebMachine, 2023-05-04:
> I'm still curious if it's the same issue. It's so intermittent it's possible that the display isn't the issue (I've done full prints just fine, and had prints fail within 10 minutes), but I'll be [...]

Given the different CPU and compile time options, a new issue would be best to start. If it turns out to be on the display firmware side, we can always merge the issues into one again.

closed 2023-07-15 by github-actions[bot]