mriscoc/Ender3V2S1 #884: Filament change problem  M125

Cowlick09, 2023-04-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

i tested this code at the very beginning of the program to verify it would work.

30 to 45 minutes into the print is my filament change on the first layer.

M125 X0 Y0 Z5 L0 P
at this line my machine moves up in z then homes x and y.  I then tap the button to continue then the screen and knob no longer function correctly.  see videos.

https://user-images.githubusercontent.com/130009909/233855532-957a0d37-ef04-419f-a340-567cddf1305d.mp4

https://user-images.githubusercontent.com/130009909/233855552-67051bba-ddc8-4dd9-a732-23b1f2721fbf.mp4

### Bug Timeline

first time i've had this issue

### Expected behavior

i expected to purge the filament and continue the print.

### Actual behavior

see Bug Description.

### Steps to Reproduce

[08GPETG_sewer Cap.zip](https://github.com/mriscoc/Ender3V2S1/files/11304646/08GPETG_sewer.Cap.zip)

### Version of Professional Firmware

Version: 2.1.3 MRiscoC  20230312-17:20

### Printer model

ender 3 pro

### Electronics

4.2.7

### Add-ons

cr-touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

I was worried my board was overheating so I placed pads under the feet and wired the motherboard fan to stay on all the time.  For some reason it was only running with the nozzle fan, in sync.  Also the room temperature is 70 degrees.  I'm very confused about this because I split the program manually to get past the first filament change freeze up and continued the rest of the program reteaching z and e and I turned off the mesh.  When it got to the second filament change it had no problem and is currently printing the rest of the part.

mriscoc, 2023-04-24:
I was not able to replicate it with the current precompiled version. Try with `M125` without parameters.

Cowlick09, 2023-05-04:
Problem exists on both my machines with precompiled firmware.  4.2.2 and 4.2.7 boards.  If i manually push the filament thru and don't use the purge button it will continue the job.  I will try M125 without parameters.  
Ender3V2-427-BLTUBL-MPC-20230312.bin
Ender3V2-422-BLTUBL-MPC-20230312.bin

My goal was to modify a script to display the time until next filament change and filament color needed now during the filament change.   I'm using the stl name for the color just don't have the time yet.

This is the only problem I have.  You've done an amazing job with the firmware .

Cowlick09, 2023-05-06:
M125 without parameters works.  Thank you.

closed 2023-05-06 by Cowlick09