mriscoc/Ender3V2S1 #886: M300 not working

Crenicara, 2023-04-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hi,
I'm coming from an old Jyers and i had this line : 
M300 P1000 S300
at the end of my end g-code.
I liked that beep telling me my print was done.
It doesn't seem to work with this firmware.

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20230312.bin 

### Printer model

Creality Ender 3V2

### Bed Leveling

None

### Your Slicer

Prusa Slicer

### Host Software

None

Crenicara, 2023-04-24:
With Jyers firmware, i didn't have the annoying beeps when pressing the display button, but the beep at the end of a print was ok. 
Is there a way to have the m300 command working without having the beeps when using the menus ?

mriscoc, 2023-04-24:
Currently, disabling the "enable sounds" in the menu have effect in all sounds.

classicrocker883, 2023-04-25:
I went and created a second button. leaving the sound checkbox as a general/master switch, and a second specific for the encoder beep/tick.

for example here is [a file from my repo](https://github.com/classicrocker883/MriscocProUI/blob/2023-April/Marlin/src/module/settings.cpp). I basically copied the `ui.sound_on` settings as `ui.no_tick`, went into */lcd/common/encoder.cpp** and changed `Encoder_tick()` to

```
void Encoder_tick() {
  TERN_(HAS_BEEPER, if (ui.no_tick) buzzer. Click(10));
}
```

instead of `ui.sound_on` it is replaced with `ui.no_tick`, so youre able to leave `sound_on` and play M300, but disable the beep/tick if you so choose as an extra option in the lcd control settings menu.

holgerlembke, 2023-05-28:
Might I upvote to change this behavior, too?

I like playing sound at the print-end, too. And dislike clicky display button sounds, too.

classicrocker883, 2023-05-28:
@holgerlembke I could compile u a firmware file from my repo, which did you flash that you are using?

holgerlembke, 2023-05-29:
Thank you very much for that offer.

I would prefer @mriscoc would rethink the plan. It would be a very good concept to let the M300 work and have the setting only for GUI stuff.

mriscoc, 2023-08-15:
With the latest release, you can use `M300 E1` to re-enable the sounds,write this just in a line before of the other M300.

closed 2023-09-11 by mriscoc