mriscoc/Ender3V2S1 #889: [Thermal Runaway] Nozzle not Heating up sometimes

Niklas344, 2023-04-25:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I am using the precompiled version [Ender3V2-SKRME3V3-BLTUBL-MPC-20230312.bin] that is the latest updated version.
I came across the Issue that my Nozzle is not heating up sometimes. The Issue appeared after 2 small Test prints < 1 hour. 
(Right after i installed the Firmware)
And since then it kills my printer at random times. 1 Time after 7 hours of printing, i guess my Nozzle temperature fall and didnt rise again. 
Now i wanted to test it again, i printed a part 28min and had no problem. After this i wanted print something other again and my Nozzle Temperature stays at 28 °C and doesnt rise. ---> leads to Thermal Runaway Issue and Killing Printer, or to Nozzle not Heating up and killing Printer

### Bug Timeline

Right after i installed the Firmware after 2 Testprints

### Expected behavior

I expect that my nozzle is heating up normally like it should and staying at the given Temperature. 

### Actual behavior

Temperature sometimes falls mid print and then killing my printer, or not even heating up sometimes and therefore killing my printer.

### Steps to Reproduce

1. Use Ender 3-V2 with SKR mini E3 V3
2. Install Spider Hotend with custom Hotend Case and direct Extruder
3. Noctua Fan on Hotend Cooling and 4010 Radial Fans for Part Cooling

spider Hotend: https://store.creality.com/de/products/spider-hochtemperatur-und-hochgeschwindigkeits-hotend?awc=32419_1682430642_ec3a1c158f9f8d58f2d7d0a9da18e5ed

Custom Case: https://thangs.com/designer/TheBuildBay/3d-model/Dual%20Blower%20Fan%20Duct%20for%20Ender%203%20V2%20with%20Spider%20Hotend%20V1%20-776965

### Version of Professional Firmware

Ender3V2-SKRME3V3-BLTUBL-MPC-20230312.bin

### Printer model

Ender 3V2

### Electronics

SKR Mini E3 V3

### Add-ons

Noctua Fan

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

ajaybnl, 2023-05-11:
I am having this too. Whenever i try to print something, the thermal runaway happens. 

Please provide a solution. I have done MPC Autotune, the problem still exists.

mriscoc, 2023-05-11:
Custom Hotends or some non-stock fans require custom build or disable thermal runaway protection.

ernestp, 2023-07-14:
Also having this issue after updating stock Ender 3 v2, no mods on hotend or fans

Never seen such drops before

[image]

[image]

ernestp, 2023-07-19:
For graphs above after more testing it seems the issue was related to testing of printing on higher speed, and spikes happens when nozzle hit deformed material on the way. So it creates tension on stepper motors, causing voltage drops which affect reading of temperature sensors.

closed 2023-09-11 by mriscoc