mriscoc/Ender3V2S1 #894: [BUG] Crash after removing SD card

qwazwsx, 2023-05-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Version: Professional Firmware v2.1.3 - Build Time 20221222 - 2:12

Firmware crashes (freezes and shows version popup) when selecting the "media" tab after printing and removing the SD card.

### Expected behavior

Expected Behavior
The media menu should set the selection to the top item ("Back") when the media is removed. It also should not crash. It appears like this is the root cause of the bug.

### Actual behavior

The firmware crashes.

### Steps to Reproduce

Reproduction:
1. Select an item in the Media menu. Either successfully print or cancel. You should be brought back to the main menu and your selection will be saved when you return to the media menu.
2. Remove the SD card. Observe "Media Removed" M117 message
3. Enter the Media menu. Observe that the selection from before is still 'selected' even though the menu is blank.
4. Move the selector wheel, the selection should jump to the top ("Back") and the firmware will hang, before showing the version info popup that appears on startup.

### Version of Professional Firmware

Professional Firmware v2.1.3 - Build Time 20221222 - 2:12

### Printer model

Ender 3v2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

Majboor, 2023-05-01:
xpected behavior should be that when the media is removed, the selection should be set to the top item "Back" in the media menu, and the application should not crash?

qwazwsx, 2023-05-01:
Just tested on the latest [Release #20230312](https://github.com/mriscoc/Ender3V2S1/releases/tag/20230312) and can confirm that I am still experiencing this bug. I'll update the issue to reflect that.

@Majboor Yes. Opening the media menu after printing and removing the SD card causes a crash. 

I've recorded a quick (and quite shaky, my apologies) demonstration video of the bug in effect on my Ender 3 v2. You can view that here: https://streamable.com/aafdwb

Here's what the bug looks like. Moving the scroll wheel at this point causes the printer to restart.
[image]

classicrocker883, 2023-05-25:
do you have SD extender installed? try enable that setting in the configuration.h file.

mriscoc, 2023-05-25:
This bug was fixed in the latest release.

closed 2023-05-25 by mriscoc

qwazwsx, 2023-05-25:
Cheers! Thanks for continuing to support this firmware

shayenss, 2023-05-25:
> Тази грешка беше коригирана в последната версия.

I have two Ender 3 S1 models and I uploaded version Ender3S1-F1-UBL-MPC-20230522B to both. I have the same problem with the SD card. It doesn't read my files and reboots, telling me the firmware version. Ask for assistance.

mriscoc, 2023-05-25:
> I have the same problem with the SD card

Sorry, but it is not the same error, the bug listed here needs a specific procedure to trigger it. You can read from the OP post that he saw the items and printed them.

Your problem probably is your SD card.