mriscoc/Ender3V2S1 #898: [BUG] (bug summary)

variableresults, 2023-05-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The print display is not displaying the correct filename when printing. The Media screen displays filenames correctly and prints the right file, but the print progress page itself sometimes shows the last file printed, or an adjacent filename in the file list. 

This is a custom compile with UBL enabled on an F4 S1 with the TJC screen (hence the custom compile, I had to hack out the same stuff mriscoc did to get it to compile for the F4's memory space). Version 2.1.3. 

### Bug Timeline

It's always existed in this version for me, I haven't tried in prior versions

### Expected behavior

Expect it to show the correct filename. 

### Actual behavior

Tends to show an adjacent filename in the filename list, or the previously printed filename

### Steps to Reproduce

Open Media, select file to print, watch as incorrect filename is displayed in the progress screen

### Version of Professional Firmware

2.1.3

### Printer model

Ender 3 S1 F4

### Electronics

Not sure

### Add-ons

Stock

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-05-02:
> ### Did you test with a precompiled firmware?
> Yes, and the problem still exists.

Please, write what precompiled version has this bug.

variableresults, 2023-05-03:
> 

That was an error on my part filling out the form, I should have selected the opposite as I compiled my own version following the Special Configurations instructions. 

I've compiled two separate versions based on the current base version, both for the TJC screen. I suspect the bug may be with that screen's code since it appeared when I compiled for vanilla ABL as well as UBL with the configuration.h and configuration_adv.h options commented out that you listed to comment to make space for UBL's code. Both compile just fine, and the display more or less works perfectly, it's just that it doesn't always display the correct filename in the Print Status screen. 

Let me know if that is helpful.

mriscoc, 2023-05-04:
Please correct the issue title, the next version has changes in the filename title management, maybe they will fix that behaviour. I don't saw that bug but maybe it depends on how many files are in the memory card.

mriscoc, 2023-05-21:
"[BUG] (bug summary)" is not a valid or meaningful title.

closed 2023-05-21 by mriscoc