mriscoc/Ender3V2S1 #900: [BUG] Disconnections on Ender 3 v2

tystuyfzand, 2023-05-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

OctoPrint disconnects from the printer at random intervals, sometimes after the first layer, mostly about 3-7 hours into a print. Doesn't appear to be a reboot, but the printer simply never responds to a command sent.

Issue is the same as the one submitted by TheWebMachine, though may be unrelated as it's a different board/firmware version:

```
2023-05-04 16:02:50,267 - octoprint.util.comm - ERROR - Unexpected error while reading from serial port
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 3997, in _readline
    ret = self._serial.readline()
  File "/usr/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 6672, in readline
    c = self.read(1)
  File "/usr/local/lib/python3.8/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)
2023-05-04 16:02:50,273 - octoprint.util.comm - ERROR - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2023-05-04 16:02:50,276 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
```

### Bug Timeline

Appeared in first few longer prints using the firmware

### Expected behavior

Print should complete successfully without hangs/disconnects

### Actual behavior

Printer randomly disconnects, sometimes after a few minutes, sometimes after a few hours.

### Steps to Reproduce

1. Start print (usually 7-8 hours are the ones I've seen it happen on most)
2. Let print continue, no settings changes or anything unexpected
3. Printer will eventually disconnect/fail communication

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20230312

### Printer model

Creality Ender 3 V2

### Electronics

4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

TheWebMachine, 2023-05-05:
When it disconnects, what does it take to recover/reconnect?
[  ] Unplug/replug USB
[  ] Restart Octoprint
[  ] Fully reboot printer

(As an aside, in my issue, it hangs and the printer reboots on its own with a full proper init. Octoprint is then able to reconnect without any other physical intervention.)

tystuyfzand, 2023-05-05:
I've actually only caught the aftermath of it once, since the PSU Control plugin turns it off after being disconnected for a while.

The printer acts as if nothing happens and stays on the "Printing" screen, but cools down/turns off eventually.

To recover, I'm usually able to just start it and connect immediately - but that means a full reboot. The one time I caught it I believe I was able to reconnect directly after, though I did press "Stop" on the printer screen before that.

TheWebMachine, 2023-05-05:
Gotcha. And is your +5 VDC line on the USB cable isolated from the printer? 

I'm told, tho I've never experienced it, that interference/instability on the USB VDC line can cause issues with hangs/disconnects/reboots.

You can butcher a USB cable or buy an isolator like this:
https://www.amazon.com/dp/B094FYL9QT
It cuts off only the +5 VDC line while keeping ground and data connected.

tystuyfzand, 2023-05-05:
I was just about to edit my last response to add that and you got me just before - it is and always has been isolated on the printer. I actually ordered that exact item as well as a decent quality Micro USB cable with ferrite core on it just to give it a bit of an upgrade in "niceness" level, because the current is a spliced and soldered cable.

closed 2023-07-15 by github-actions[bot]