mriscoc/Ender3V2S1 #902: [BUG] Printer crashes ("Printer killed") during 9x9 ABL

al7810, 2023-05-06:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

My Printer crashes during 9x9 ABL and displays the message "Printer killed". Motherboard with STM32F401RCT6 SoC.
ABL with 8x8 mesh works as expected. Perhaps an out-of-memory issue?

This behaviour occures with your (fantastic) Boreal Spring 2023 release (Ender3S1-F4-20230312) and also with my self compiled firmware with TJC support.

Thank you for all the work you do.

### Expected behavior

9x9x ABL should work as expected or not beeing allowed e.g. in case of insufficient memory

### Actual behavior

Printer crashes during ABL with message "Printer killed"

### Steps to Reproduce

Try to perform ABL with 9x9 mesh

### Version of Professional Firmware

Ender3S1-F4-20230312 and self compiled version with TJC support

### Printer model

Ender 3 S1

### Electronics

STM32F401RCT6

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-05-06:
HS?

al7810, 2023-05-07:
HS = Host Software? Currently I'm using a SD Card (headless) only. But I can setup octopi if it helps.

TheWebMachine, 2023-05-07:
I think he means, "are you using the High Speed Probe option?"

al7810, 2023-05-07:
I love abbreviations :-) Disabling High Speed Probe option resolves the problem. 9x9 ABL mesh can be build w/o "killing the printer"

TheWebMachine, 2023-05-07:
> I love abbreviations :-) Disabling High Speed Probe option resolves the problem. 9x9 ABL mesh can be build w/o "killing the printer"

Haha Yeah, shorthand is only handy when everyone knows the lingo. Otherwise, it just adds more confusion than it resolves. 

So, your probe and/or board are unstable in HS mode, but that's not a big deal. Non-HS mode is just a little slower since it stows and redeploys the probe every stop. All else should be fine.

Glad we could sort that out! 😎

mriscoc, 2023-05-08:
It is always a good practice to read the documentation.

https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

closed 2023-05-08 by mriscoc