mriscoc/Ender3V2S1 #911: [BUG] (bug summary)

m0nk1111, 2023-05-12:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

if i pause the print to change filament, the head moves to an outer corner, where a clip is to hold the buildplate in its place.
it gets stuck on the clip, but it thinks it still moving, which results in the coordinates being off. so when i resume it resumes at wrong coordinates. 

can the hotend plz not move to the corner, but in the middle, where no buildplate clip is to be expected?

### Bug Timeline

i just noticed it

### Expected behavior

when pausing print, move the head to a location, where no buildplate hold in place clips would be expected

### Actual behavior

it moves to my clip in the right back corner, when i hit pause print

### Steps to Reproduce

oke

### Version of Professional Firmware

the last one

### Printer model

CE3V2

### Electronics

4.2.2

### Add-ons

non

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-05-12:
Check your park head position
https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

m0nk1111, 2023-05-12:
oke then let me put it differently
why is the park head default position in the right corner for an ender 3 Vx when theres clips in the corners most of the time?

mriscoc, 2023-05-12:
The default park position is calibrated for an all-stock Ender 3 V2 and allows for post-purge nozzle cleaning when a break or filament change is required.

m0nk1111, 2023-05-12:
oke
but do you reconize my problem here?

mriscoc, 2023-05-12:
Do you have a totally stock Ender 3V2?
The defaults settings should work.

m0nk1111, 2023-05-12:
i have a second z axis and some led lighting,
[image]

and some custom clips

m0nk1111, 2023-05-12:
oh, running prints from octoprint could be related
i start al my prints from octoprint
but i paused it with the UI on the printer

closed 2023-07-12 by m0nk1111