mriscoc/Ender3V2S1 #912: [BUG] Extruder fan not coming on automatically as set in slicer

KobraNeo, 2023-05-13:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I will initiate a print with the extruder fan disabled for the first 1-3 layers depending on the material/project, and when the layer is reached for the fan to  kick on the display will show the speed it should be at, but the fan wont be moving. I will go in and manually set the fan speed to just a few units lower, itll start, and then I can turn it back up to full speed. 

### Bug Timeline

I noticed May 12, 2023 but print quality has been compromised for a week

### Expected behavior

I expected the fan to come on at the desired setting at the desired layer

### Actual behavior

The fan did not turn on until I set it to a lower value manually and then turned it up

### Steps to Reproduce

1. Slice Model
2. Start print
3. Fan doesnt start

### Version of Professional Firmware

Ender3S1-F4-UBL-20230312

### Printer model

Creality Ender 3 S1 F4

### Electronics

TJC display- Would love update for that for mesh display

### Add-ons

5015 fan with new duct

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

SD Card (headless)

### Additional information & file uploads

No additional mods made besides extruder cooling and accompanying duct. Polarity of fan was ensured at install, and the fact it will turn on shows some error is happening with the initialize action. 

I would also love the FBL display update to view my mesh values; I know that was only available through a chat service I dont use. 

Thank you in advance, and once again I appreciate what youre doing to improve these cheap chinese bedslingers.

mriscoc, 2023-05-14:
Precompiled firmware is compatible with stocks fans and fans that use the same PWM signal, others fans could be incompatible.

KobraNeo, 2023-05-21:
I noticed yesterday that on this F4 T13 firmware that the light stays lit on my ABL probe. Should that be a concern?

mriscoc, 2023-05-22:
> I noticed yesterday that on this F4 T13 firmware that the light stays lit on my ABL probe. Should that be a concern?

BLTouch stays with a red light.

KobraNeo, 2023-05-23:
Mine is staying purple; the blue and red are simultaneously on

closed 2023-08-01 by github-actions[bot]