mriscoc/Ender3V2S1 #928: [QUESTION] Is the customizable color menu missed in the latest update? I'm on F4 UBL

UsSuggester, 2023-05-24:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Asking because I don't see it as a missing option due to lack of memory on F4. I was really a fan specially of dark background but, if the printer works better without it, welcome.

### Version of Professional Firmware

Ender3S1-F4-UBL- 2023 05 22 B

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-05-24:
Yes, I removed it by disabling `HAS_CUSTOM_COLORS` in the configuration file to make more program memory space for UBL.

UsSuggester, 2023-05-24:
Thank you. Solved.

closed 2023-05-24 by UsSuggester