mriscoc/Ender3V2S1 #936: [BUG] Mesh - Auto level with - CR Touch Ender 3 v2

llopepe, 2023-05-29:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hello, I have been using it for a while in my ender 3 v2, this latest version is available and something strange happened to me with the cr touch when the 25-point bed is made, I noticed that it only performs a single verification, the previous version 20230312 performs 3 verifications at each point. The bed at the end gave me very bad (between -0.18 to 0.01), but I changed to the previous version, reconfigured everything and it gave me a better result (-0.03 to 0.01).

**version 20230522**

[image]

**version 20230312**

[image]

### Bug Timeline

new bug 

### Version of Professional Firmware

Mesh - Auto level with

### Printer model

Creality Ender 3 v2

### Electronics

4.2.2

### Add-ons

cr touch

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-05-29:
Hello. You tested the surface of your bed with a ruler, the second image looks unusually flat.
You can add more point measurements by increasing the Multiprobing in the probing settings menu.

leolopepe-dawa, 2023-05-29:
Hola muchas gracias, por tu respuesta tan rápida, hablando con un amigo que te seguimos me dice, podes escribirle en español!!!  Como vi que todos escribían en ingles!!! Bueno...

Voy a hacer eso que me comentas, de probar esa opción de configuracion o dejarlo asi como esta, si la nivelación es mucho mas rápida en este ultima version. Muy buen trabajo, saludos!!!

llopepe, 2023-05-29:
Hola muchas gracias, por tu respuesta tan rápida, hablando con un amigo que te seguimos me dice, podes escribirle en español!!! Como vi que todos escribían en ingles!!! Bueno...

Voy a hacer eso que me comentas, de probar esa opción de configuracion o dejarlo asi como esta, si la nivelación es mucho mas rápida en este ultima version. Muy buen trabajo, saludos!!!

closed 2023-05-29 by llopepe