mriscoc/Ender3V2S1 #949: [Stops feeding when I adjust z probe offset from print] (Basically just prints in same spot over and over after I tune z probe offset)

Aspenthedon, 2023-06-03:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

All of the sudden now on this print when I tune z probe offset while printing the printer seems to print in place and stops feeding pla. Any other time I've tuned z probe offset there hasn't been any issue. could this be a issue with slicer settings? I'm lost and would love some guidance! Thanks

### Bug Timeline

6-3-23

### Actual behavior

expected for print to continue as normal after adjusting z probe offset

### Steps to Reproduce

start printing
tune z probe offset
then printer stops feeding and just prints same line over and over 

### Version of Professional Firmware

Ender-3 V2/S1 Professional Firmware 20230312

### Printer model

ender 3 v2 neo

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-06-03:
Change SD card.

Aspenthedon, 2023-06-04:
ok I will do that and report back. Thanks

Aspenthedon, 2023-06-04:
So I changed my sd card out with well known brand sd and still same issue. Should I try to update firmware to latest release? Thanks for you quick response man!

Aspenthedon, 2023-06-04:
Also I'm printing through octoprint if that make a difference. I thought you should know that detail.

mriscoc, 2023-06-04:
If you print from Octoprint, then it's not a problem with the SD card, it's a communication problem with your Octoprint. Try saving the G-code file in the SD card and print from it to discard a problem with the slicer.

Aspenthedon, 2023-06-05:
so I printed same print from new sd card and same issue happens. I forgot to mention that the printer also does eventually start printing again but it causes issues with the print, like you can see where it has frozen and then started printing again because it leave a ridge in the pla. any suggestions on what I should do? Nothing has changed since I update firmware and everything was working perfectly and all of the sudden this happened.

mriscoc, 2023-06-05:
Return to a previous working firmware and try with the exact same G-code file. Is the pause always in the same position?

Aspenthedon, 2023-07-29:
I have went to previous working version of your fw. Did you mean revert to stock fw? Yes it is always in same spot. I have tried to reslice and still same issue. I have upgraded your fw with same issue. I cannot figure out this issue. 

I don't know if I mentioned that after it runs in same spot with pla not feeding for some time it will resume print and then after a bit do it again. I havr attached a image if this is of any help. Thanks man
[image]

mriscoc, 2023-07-29:
What I tried to say is, identify the problematic g-code file, do not re-slice it. Flash a firmware version that you know worked before and try to print the exactly same g-code file.

closed 2023-09-11 by mriscoc