mriscoc/Ender3V2S1 #953: [BUG] FW makes 2 writes to LCD when configured with TJC_DISPLAY

mill1000, 2023-06-06:
### Did you test with a precompiled firmware?

No, but I will test it now!

### Bug Description

If the FW is configured for a TJC screen (i.e. TJC_DISPLAY is defined), the following function will make a call to both subfunctions.

https://github.com/mriscoc/Ender3V2S1/blob/3a06db5ba492ea8819fdc7cf154fc7d2c1a88c69/Marlin/src/lcd/e3v2/proui/dwin_lcd.cpp#L133-L140

I expect only a single subfunction should be called. Here's a possible fix which utilizes the `DWIN_ICON_Show` function for TJC_DISPLAY.
```c++
void DWIN_ICON_Show(uint16_t x, uint16_t y, uint16_t addr) {
  #if ENABLED(DACAI_DISPLAY)
    DACAI_ICON_Show(x, y, addr);
  #elif ENABLED(DWIN_DISPLAY) || ENABLED(TJC_DISPLAY)
    DWIN_ICON_Show(0, 0, 1, x, y, addr);
  #else
    #error "No DWIN compatible display enabled"
  #endif
}
```

I tried both functions on my TJC screen and saw no changes in behavior (Unfortunately! I was hoping to fix the icon transparency)

### Version of Professional Firmware

20230522

### Printer model

Ender 3 V2 Neo

### Electronics

Board 4.2.2

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-06-06:
The original function is a little trick that works when no display was predefined.

closed 2023-06-06 by mriscoc

mill1000, 2023-06-06:
Ok, but what about when a TJC screen is defined? No need to keep both around for that right?

mill1000, 2023-06-07:
Yes it does work with my TJC screen. But it could still be improved.