mriscoc/Ender3V2S1 #957: [BUG] Custom G-code C104 is not working in latest 20230522 version

mriscoc, 2023-06-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Reported by @Mobius135 at [reddit](https://www.reddit.com/r/Ender3V2NEO/comments/13r8ral/comment/jng73ti/?utm_source=reddit&utm_medium=web2x&context=3): 

```log
Send: C104 
Recv: echo:; Max Extruder temperature: 
Recv: echo: C104 T275 
Recv: ok P15 B15 
Send: C104 U1 T275 
Recv: Error: Thermistor max: 0 
Changing monitoring state from "Operational" to "Error"
```

Setting the maximum temperature with C104 gives an error response:
`Recv: Error: Thermistor max: 0`

### Bug Timeline

New bug

### Expected behavior

C104 must allow to set the maximum temperature in the working range of the thermistor.

### Actual behavior

Setting the maximum temperature with C104 gives an error response:
`Recv: Error: Thermistor max: 0`

### Steps to Reproduce

Sending C104 custom G-code using Octoprint gives:

```log
Send: C104 
Recv: echo:; Max Extruder temperature: 
Recv: echo: C104 T275 
Recv: ok P15 B15 
Send: C104 U1 T275 
Recv: Error: Thermistor max: 0 
Changing monitoring state from "Operational" to "Error"
```

### Version of Professional Firmware

20230522

### Bed Leveling

None

### Your Slicer

None

### Host Software

OctoPrint

mriscoc, 2023-06-09:
The error was caused by a bug in Marlin: https://github.com/MarlinFirmware/Marlin/issues/25953

ellensp, 2023-06-09:
Which has already been fixed (by you) and has been merged, so this is fixed?

mriscoc, 2023-06-09:
> Which has already been fixed (by you) and has been merged, so this is fixed?

Only in the source code, obviously the precompiled binaries still have the bug.

KBalage, 2023-06-26:
It doesn't work for me either - I have an upgraded extruder, should be able to cope with 290, but receive this:

Send: C104 U1 T290
Recv: Error: Thermistor max: 0
Recv: echo:; Max Extruder temperature:
Recv: echo:  C104 T275

codecompliant, 2023-07-08:
Any idea when this will be fixed?

mriscoc, 2023-07-31:
> Any idea when this will be fixed?

It was already fixed in the source code, so you can compile or request a custom build or wait to the next august release.

closed 2023-09-11 by mriscoc