mriscoc/Ender3V2S1 #958: [BUG] typo in feature/mmu/mmu2.cpp

classicrocker883, 2023-06-09:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Line 143 says `HMI_data.Runout_active_state`

it should be `PRO_data.`

### Version of Professional Firmware

Newest release - May 20230522

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

while I post this here. I would also like to post this

```
#if ENABLED(DWIN_LCD_PROUI)
// Professional firmware features:
  //#define PROUI_EX 1            // Extended UI features (15152 bytes of flash)
  #ifdef PROUI_EX
    #define HAS_GCODE_PREVIEW 1
    #define HAS_TOOLBAR 1
  #endif
  #define HAS_PLOT 1            // Graph Temp as grid plot - PID/MPC Tuning (1624 bytes of flash)
  #define HAS_ESDIAG 1          // View End-stop switch continuity (560 bytes of flash)
  #define HAS_CGCODE 1          // Extra Gcode options (3320 bytes of flash)
  #define HAS_LOCKSCREEN 1      // Simple lockscreen as to not accidentally change something (568 bytes of flash)
  #define HAS_SD_EXTENDER 1     // Enable to support SD card extender cables (48 bytes of flash)
  #define USE_GRID_MESHVIEWER 1 // Enable two mesh graph types : one (1728 bytes of flash)
  #define HAS_CUSTOM_COLORS 1   // Able to change display colors (2040 bytes of flash)
  #define ALTCOLOR_MENU_ITEM 0  // Color palette options => Disabled or 0 = Voxelab Default | 1 = Alternate Aquila | 2 = Ender3V2 Default
  #if ENABLED(AUTO_BED_LEVELING_UBL)
    #define ACTIVATE_MESH_ITEM  // Active Mesh Leveling menu option (152 bytes of flash)
  #endif
  #if ENABLED(FILAMENT_RUNOUT_SENSOR)
    #define RUNOUT_TUNE_ITEM    // Filament Runout option in Tune Menu (2528 bytes of flash)
  #endif
  #if ENABLED(POWER_LOSS_RECOVERY)
    #define PLR_TUNE_ITEM       // Power-loss Recovery option in Tune Menu (3400 bytes of flash)
  #endif
  #if ENABLED(BLTOUCH)
    #define HS_MENU_ITEM        // BLTOUCH_HS_MODE menu option (56 bytes of flash)
  #endif
  #if ENABLED(HAS_PLOT)
    #define PLOT_TUNE_ITEM      // Temperature Plot Graph item in Tune Menu (688 bytes of flash)
  #endif
  #if DISABLED(CLASSIC_JERK)
    //#define JD_TUNE_ITEM      // Enable only if Juntion Deviation is enabled
  #endif
  #if ENABLED(LIN_ADVANCE)
    #define ADVK_TUNE_ITEM      // Linear Advance item in Tune Menu
  #endif
  #define SHOW_REAL_POS
  #define CCLOUD_PRINT_SUPPORT  // Menu item: enable/disable Creality Cloud Print Support (192 bytes of flash)
  #define TRAMWIZ_MENU_ITEM     // Menu item: enable Tramming Wizard (2304 bytes of flash)
  #define MEDIASORT_MENU_ITEM   // Menu item: enable/disable file list sorting (104 bytes of flash)
  #define ENC_MENU_ITEM         // Menu item: faster/slower encoder rate (272 bytes of flash)
  #define SHOW_SPEED_IND        // Menu item: blink speed in mm/s along with speed percentage (296 bytes of flash)
  //#define NO_BLINK_IND        // Disables dashboard icon blink indicator highlighted background
  #define ZHOME_BEFORE_LEVELING

#endif

```

I went ahead and defined how many bytes each option uses when enabled. I hope this is useful.

mriscoc, 2023-06-11:
Thanks, Runout_active_state was part of HMI_data on early versions and then moved to PRO_data, but that particular file was not changed. I don't know if currently PROUI_EX will work with MMU enable printers.

mriscoc, 2023-06-11:
> I went ahead and defined how many bytes each option uses when enabled. I hope this is useful.

That is very helpful info, it might be interesting to put it in the [Discussions section](https://github.com/mriscoc/Ender3V2S1/discussions).

Crazy-Charles, 2023-06-11:
pause.cpp also has the same typo

classicrocker883, 2023-06-11:
of course just this one menu item doesn't take this much, I meant for having `FILAMENT_RUNOUT_SENSOR` enabled uses this much memory. I suppose moving the **(2528 bytes of flash)** to the actual line of `#define FILAMENT_RUNOUT_SENSOR` makes sense, clarification might be needed.

this can be reworded to

```
#if ENABLED(FILAMENT_RUNOUT_SENSOR) // (2528 bytes of flash)
  #define RUNOUT_TUNE_ITEM    // Filament Runout option in Tune Menu
#endif
#if ENABLED(POWER_LOSS_RECOVERY) // (3400 bytes of flash)
  #define PLR_TUNE_ITEM       // Power-loss Recovery option in Tune Menu
#endif

classicrocker883, 2023-06-15:
there are several more typos such as in **proui/dwin_popup.h**
`inline void Draw_Select_Highlight(const bool sel) { Draw_Select_Highlight(sel, 280); }; `

<br>

and in **proui/dwin.h**
`inline void DWIN_Gcode(const int16_t codenum) { TERN_(HAS_CGCODE, custom_gcode(codenum)); };`

as you can see at the end ";}&nbsp;<i>  **;**</i>"

there are a few more I can make a PR, this isn't a big problem but it doesn't hurt. also { } should be added to any code for '**if**' and '**else**' statements, which there are a lot missing and that can cause some issues.

closed 2023-08-25 by github-actions[bot]