mriscoc/Ender3V2S1 #963: Filament Runout Sensor not working

Ozrensky, 2023-06-11:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I have a problem with a runout sensor. Standard S1 F1 board.

HIGH => Endstop Test is always TRIGGERED no matter if filament is loaded or not. During a print it never stops after the filament has run out from the sensor, it starts a print even if filament is not loaded at all.
LOW => Endstop Test is always OPEN. During print it pauses constantly and asks for a filament load.

Tried precompiled Ender3S1-F1-UBL-IS-MPC-20230524 as well as configurator compiled 20230524 firmware.

### Bug Timeline

This is the first time I installed this firmware, previously I had stock firmware.

### Expected behavior

Runout is expected to get triggered when the runout sensor detects it.

### Actual behavior

Endstop Test for the Runout sensor is always TRIGGERED no matter if filament is loaded or not (I guess it is not reading data from the sensor at all). During a print it never stops after the filament has run out from the sensor.

### Steps to Reproduce

Simple, try putting filament in and out when the printer is not printing and check Endstop Test.

Try printing without loaded filament with LOW and HIGH set.

### Version of Professional Firmware

2.1.3

### Printer model

S1 F1

### Add-ons

None

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

mriscoc, 2023-06-12:
Please check your hardware, the filament runout sensor has been reported to work as expected with the current version.

mriscoc, 2023-06-12:
If it works perfectly with the original firmware and the HIGH/LOW setting does not work for you, then this firmware version may not be compatible with your hardware.

Ozrensky, 2023-06-12:
What can I do to fix this? In Configurator printer is S1, board is F1, screen is DACAI, maybe thermistor is wrong? Sorry for waisting your time, I just want to find a solution.

mriscoc, 2023-06-13:
The filament runout sensor can be configured in the filament settings submenu. It can also be tested using the endstops diagnostic tool in the Advanced settings menu. If neither configuration works with your filament sensor, there may be a currently unresolved incompatibility for your specific hardware configuration.

Ozrensky, 2023-06-13:
Ok, so I've installed the precompiled F1 firmware from the previous release and the filament runout sensor works. It just has to be a bug with the latest release.

mriscoc, 2023-06-13:
> Ok, so I've installed the precompiled F1 firmware from the previous release and the filament runout sensor works. It just has to be a bug with the latest release.

That's a good test, but it is weird because the runout routine was tested and it was working before the release. It is working in my printer, a V2, so maybe it could be a problem with sensor pin definition.

mriscoc, 2023-06-13:
The printer will stop printing only after the extruder moves the "runout distance", for optimize the use of the filament between the sensor and the extruder. Currently in the Ender-3 S1, the runout distance is 200 mm (20 cm) and it is configurable in the filament settings menu.

mriscoc, 2023-06-13:
What happen if Runout active is set to HIGH with filament in the sensor (TRIGGERED) and print?

mriscoc, 2023-06-13:
Thanks for your tests, I need to dig deeper to check how the sensor signal is being interpreted.

mriscoc, 2023-06-13:
Could you test a simple version as [Ender3S1-F1-UBL-MPC-20230522B.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230522/Ender3S1-F1-UBL-MPC-20230522B.bin)?

Ozrensky, 2023-06-13:
Print is running at the moment, so I can try it in about 12 hours from now. I'll write my findings here, thanks.

Ozrensky, 2023-06-14:
> Could you test a simple version as [Ender3S1-F1-UBL-MPC-20230522B.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230522/Ender3S1-F1-UBL-MPC-20230522B.bin)?

Runout sensor is detecting when filament run out during print and pauses (working as intended).

It still doesn't trigger if there is no filament in the sensor when u start the print. If you put the filament in and out (after the print started without filament) it will detect it, and after that it will detect every runout. (I could live with that but I am telling you just in case)

mriscoc, 2023-06-14:
> Runout sensor is detecting when filament run out during print and pauses (working as intended).

So, there is something in the IS code that prevents the run-out routines from working as intended.

> It still doesn't trigger if there is no filament in the sensor when u start the print.

Run-out routines are triggered by extruder movements, so the printer will not perform advanced pause until the first extrusion movement.

Ozrensky, 2023-06-14:
At least we have a clue now. Are you going to take a look at IS or is it going to stay as it is?

mriscoc, 2023-06-14:
> At least we have a clue now. Are you going to take a look at IS or is it going to stay as it is?

I'll check if there's anything obvious that needs to be fixed, but Marlin is changing the IS code because it's a new feature. If I make some change, it could be invalid for the next IS modification. I prefer to wait to have a more stable IS code.

Ozrensky, 2023-06-14:
Alright, thanks a lot!

mriscoc, 2023-06-14:
All the binaries that I posted here to your tests, were building using the current published source code. So it is weird the different behavior when some features are enabled.

classicrocker883, 2023-06-14:
what if you Change the runout pull[up]/[down] setting? `FIL_RUNOUT_PULLUP` to `FIL_RUNOUT_PULLDOWN`?

Ozrensky, 2023-06-15:
> what if you Change the runout pull[up]/[down] setting? `FIL_RUNOUT_PULLUP` to `FIL_RUNOUT_PULLDOWN`?
> 
> (or down to up)

I've just installed pro firmware like 2 weeks ago, can you tell me in which script should I change that?

classicrocker883, 2023-06-15:
thats in the **Marlin/Configuration.h** file

look for `#definfe FILAMENT_RUNOUT_SENSOR` and youll see the pullup/pulldown option, choose one to be enabled.

also check out these files which I recently updated with the newest Marlin - if youre able to copy and paste the code and compile it may fix the issue.

[runout.h](https://github.com/classicrocker883/MriscocProUI/blob/2023-June/Marlin/src/feature/runout.h) on my github repo
&nbsp; &nbsp; [raw page](https://raw.githubusercontent.com/classicrocker883/MriscocProUI/2023-June/Marlin/src/feature/runout.h) of *runout.h*

<br>

[runout.cpp](https://github.com/classicrocker883/MriscocProUI/blob/2023-June/Marlin/src/feature/runout.cpp) on my github repo
&nbsp; &nbsp; [raw page](https://raw.githubusercontent.com/classicrocker883/MriscocProUI/2023-June/Marlin/src/feature/runout.cpp) of *runout.cpp*

Ozrensky, 2023-06-15:
So I changed from pulldown to pullup and pasted your runout.h and runout.cpp. In the diag screen it shows correctly open/triggered, but during the print it doesn't trigger runout sensor

classicrocker883, 2023-06-17:
you have to enable it from the menu. the runout enabled checkbox did u do that? and it should be set to High or low, one will trigger without filament detected.

Ozrensky, 2023-06-17:
of course I did that

classicrocker883, 2023-06-17:
I'm looking through the code there may be something I didn't include

classicrocker883, 2023-06-17:
https://raw.githubusercontent.com/classicrocker883/MriscocProUI/5d030c6b5cef233fa73a05ff848a96420810389c/Marlin/src/feature/runout.h

try this code/file

Ozrensky, 2023-06-23:
Tried it, still the same behaviour. But it definitely works without IS in build.

Ozrensky, 2023-06-23:
After this try, no matter what feature I include in build, runout sequence won't trigger during the print.. I've downloaded fresh special configurations as well as the main repository and still it doesn't work. I am lost at this point 😅

classicrocker883, 2023-06-23:
have you tried without the pro features? with ( `PROUI_EX` ) disabled.

Ozrensky, 2023-06-24:
I've disabled it (with 0 instead of 1 in Confiiguration.h) but it fails to build

Ozrensky, 2023-06-25:
Ok so I've tried a lot of things, reverted to a couple of older versions and tried to make a build with filament runout working but with no success. 
Configuration => Ender3S1, 301F1, BLT + UBL, DACAI, T1, any feature
What are my findings:
- it won't work even with the older versions (I've reverted both firmware and configuration so there is no mistake with that)
- build with up to date fimware (09/06/23 commit) and up to date configurator (30/05/23) => end stop diag showing correct values TRIGGERED/open, but during the print runout won't trigger (HIGH runout active)
- it doesn't matter which features I select, outcome is the same, so we can't say it is because of the IS
- only precompiled build that I have tested that actualy works is [Ender3S1-F1-UBL-MPC-20230522B.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230522/Ender3S1-F1-UBL-MPC-20230522B.bin)
- precompiled latest (for example) LA build for the S1 F1 is not working correctly as well
- Creality firmware for the S1 F1 is working correctly

I won't waste your time with this issue anymore because it seems that there is no solution for it at the moment. I just hope there will be in the near future because I am just missing a printer feature now. 
Thanks for the assistance.

classicrocker883, 2023-06-26:
you have to comment out PROUI_EX - sorry thats what I meant by disabling.

so its like 
> // #define PROUI_EX 1

Ozrensky, 2023-06-26:
Tried that just now and still the same symptoms... It's probably something with how runout scripts work.. I tried to compare them with creality's runout scripts and there are some differences with methods, but I was unable to get them to work inside this firmware. I am really not into this kind of development so I that's the best I can do.

classicrocker883, 2023-06-26:
can you show me the scripts you're talking about?

so the filament sensor just doesn't work with Input Shaping enabled?

then I would assume it has more to do with the input shaping code than runout. there must be a conflict. I wonder if MarlinFirmware has any idea. might be beneficial to post an issue there if you haven't already

Ozrensky, 2023-06-26:
Runout sensor is not working with the latest firmware and configurator (at least), no matter which feature I select, even if I make a build without any feature it won't work.. so it is definitely not because of the IS alone. 
I found creality scripts from creality s1 repo: https://github.com/CrealityOfficial/Ender-3S1

closed 2023-06-29 by Ozrensky

mriscoc, 2023-08-22:
@shaulbarlev did you check https://github.com/mriscoc/Ender3V2S1/issues/1025?

shaulbarlev, 2023-08-24:
Yes I checked #1025 after typing the last message. Setting it to LOW fixed it for a couple of prints but now the problem is back. I switched from a non UBL build and back and maybe then it showed again. I made sure to set it to LOW back again after the reflashing and the problem persists.

mriscoc, 2023-08-24:
> I made sure to set it to LOW back again after the reflashing and the problem persists.

Don't forget to save settings. If you want to be more sure, add the proper gcode to the slicer start script.

```gcode
C412 Mn ; set run-out sensor active state (M0:LOW, M1:HIGH, M2:MOTION)
or
M412 S0 ; disable filament runout sensor 
```

shaulbarlev, 2023-08-25:
@mriscoc I now tried M412 in the print gcode and set it to LOW, the way it worked before - but the problem persists. Thank you