mriscoc/Ender3V2S1 #964: Invlaid Mesh Error when Mesh size is not 7x7 [BUG] (bug summary)

geeffland, 2023-06-15:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

If the saved mesh is not 7x7 the latest (20230533) will show a "Invalid Mesh" error when turning on the printer.  It looks like the mesh size reverts to 7x7 upon startup and won't load the mesh.  In my case I used a 5x5 and if I change the mesh size to 5x5 before loading the mesh it will work.

Ender 3 V2 with CR Touch -- Firmware = Ender3V2-422-BLTUBL-MPC-20230522B

### Bug Timeline

First noticed with this release.

### Expected behavior

I expect to be able to use any valid mesh size and have it load when starting the printer

### Actual behavior

If the saved mesh is different than 7x7 (5x5 in my case), the printer shows "invalid mesh" when turned on, or when load mesh is used after turning on.  Also notice that the mesh size is back to 7x7 instead of the previous mesh size.

### Steps to Reproduce

1. Create 5x5 bed leveling mesh
2. Save in Slot 0
3. Turn off printer
4. wait
5. turn back on
6. see error message... or if you don't catch it go to advanced -> bed leveling menu note that size is now 7x7, try a load mesh and get error

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20230522B

### Printer model

Ender 3 V2

### Electronics

Board 4.2.2

### Add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-06-15:
You forgot to also save settings, slot number and mesh size are configuration parameters, so they must be saved to be persistent after reboot.

classicrocker883, 2023-06-16:
have you tried resetting the mesh (zero out all values)? u can do so in the mesh edit menu. see if resetting the mesh after setting to 5x5. then save it. u may need to reload it after restart.

geeffland, 2023-06-18:
I have confirmed that if the mesh size is changed and "Store Settings" is used then the mesh size remains on the next power on, and the mesh loads properly.  So I agree with the "False Alarm" flag.

closed 2023-06-18 by geeffland