mriscoc/Ender3V2S1 #965: [BUG] Hotend PID error

eblieb, 2023-06-15:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Send: M303 C8 E0 S200 U1
Recv: PID Autotune failed! Bad heater id

Something is going on where it doesn't identify E0 as the hot end? 

### Version of Professional Firmware

Send: M303 C8 E0 S200 U1 Recv: PID Autotune failed! Bad heater id

### Printer model

Ender 3V2

### Electronics

4.2.2

### Add-ons

CRTouch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

mriscoc, 2023-06-15:
Maybe you are using an [MPC version](https://github.com/mriscoc/Special_Configurations/releases/tag/MPC) of the firmware, so there is no PID for the extruder.

xingjl86, 2023-08-21:
I am working with MPC version now. And also see this error message. It is acting so good, ignore PID.

mriscoc, 2023-08-21:
> I am working with MPC version now. And also see this error message. It is acting so good, ignore PID.

For MPC versions there is no PID for the hotend. MPC and PID are mutually exclusive.

closed 2023-08-21 by mriscoc