mriscoc/Ender3V2S1 #966: [BUG] Firmware cannot be installed/used with latest DACAI display firmware, screen stuck on logo

martinwilco, 2023-06-18:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I previously used the professional firmware 20230522, but then switched to the Creality firmware 3.1.9 to test the new model picture preview function.
Downloaded from: [https://www.creality.com/pages/download-ender-3-s1](https://www.creality.com/pages/download-ender-3-s1)
Creality includes a firmware update for the display with the familiar folder "private" and the file "firmware.zlib". Additionally there is a file called "dcboot.bin" that needs to be installed before flashing private and firmware.zlib.
The display works absolutely fine with Creality's firmware 3.1.9

But when flashing the professional firmware 20230522 again, there is no loading bar as usual and the printer does not boot up. It gets stuck on the Creality logo screen. Neither flashing the older firmware "dacai_update.zip" from [display assets](https://github.com/mriscoc/Ender3V2S1/tree/Ender3V2S1-Released/display%20assets) nor renaming the firmware file helps.
Older display firmware versions from Creality do not work either. A downgrade from "dcboot.bin" is not available.

### Bug Timeline

After updating display firmware

### Expected behavior

Boot up as usual, firmware flashing with loading bar

### Actual behavior

When booting up screen is stuck on Creality logo, the printer cannot be used at all.

[image]

### Steps to Reproduce

1. Install "dcboot.bin" from [Creality firmware 3.1.9](https://file2-cdn.creality.com/file/68142d4f0423807c9efe5283afdedb43/Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER.zip) on DACAI display via microSD-card
2. Flash private folder and "firmware.zlib"
3. Try to flash professional firmware 20230522, in my case [Ender3S1-F4-20230522B.bin](https://github.com/mriscoc/Ender3V2S1/releases/download/20230522/Ender3S1-F4-20230522B.bin)

### Version of Professional Firmware

20230522

### Printer model

Creality Ender-3 S1

### Electronics

V24S1_301, DACAI

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

### Additional information & file uploads

A fix would be very much appreciated.
Otherwise the only way is to buy a different display to use the professional firmware, since a firmware downgrade does not seem to be available.

Thank you so much for your effort!!

[image]

mriscoc, 2023-06-18:
Sorry but it is not a issue with this firmware, you must report in Creality's Github to request to them a way to revert their issue.

martinwilco, 2023-06-18:
Ok thanks for the info. If I am not the only one with this issue, you can maybe also add a warning to your wiki so that others do not make the same mistake:)
Then I will try my luck with Creality's support.

djcath, 2023-06-21:
im also unable to flash the firmware

Crapkin22, 2023-06-24:
I found a work around this problem. I asked creality support for older firmware. Flashed it. Then I reinstalled the mriscoc firmware. It worked. 
Here  is the link to firmware they gave me. [https://www.crealitycloud.com/software-firmware/firmware?type=2&keyword=&system=0&subType=618ddf638f5f262859ef2091&hardware=V24S1_301](url)

I used the firmware from 28.10.2022

djcath, 2023-06-24:
Was this specifically the printer firmware or was it both display and printer firmware as I plan to try this. 
I also found that the latest creality display firmware wouldn’t actually install on my S1 said the file was too large or something.

Crapkin22, 2023-06-24:
Here are the steps I did.
1. Flashed latest creality screen and motherboard firmware version 3.1.9 the one with dcboot file for the screen since the 3.2 version  didnt work on my printer.. (just to get the printer working).
2. Flash old screen and motherboard firmware, from the link I provided above (file dated 28.10.2022)
3. Flash mriscoc firmware.  everything seems  to work fine. 
Tried installing the screen firmware for mriscoc for DACAI screen but that still doesnt work (black screen).
So i went back to old creality screen firmware and it seems to work fine, Menus are all  there.

djcath, 2023-06-24:
Amazing thanks for this.    to add.  i read a bit further and saw that my display was TJC (not dacai or Dwin) based on the pics.   so i loaded the TJC update (and also put in the DACAI firmware folders too for the hell of it)   this updated my screen,  then there is a Ender S1 F4 TJC firmware,  this seemed to work (release tag is TJC)

Crapkin22, 2023-07-02:
> 

https://www.crealitycloud.com/software-firmware/firmware?type=2&keyword=&system=0&subType=618ddf638f5f262859ef2091&hardware=V24S1_301

This is the link that was provided to me by the tech support for older firmware. I used the firmware Endeer-3 S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER. its dated 28.10.2022

Crapkin22, 2023-07-02:
> Agh. I'm getting nowhere with that firmware sadly. I've tried every way I
> can think of getting back my display but it looks like it's refusing to
> actually boot into anything now.
>
> Did you definitely apply dcboot.bin to your screen?

Yes. I wrote a step by step of what i did above.

martinwilco, 2023-07-03:
> @martinwilco can you update your description to reflect the fact that a downgrade is possible?
> 
Ok so I've read everything you've done so far. My solution was to buy a different display...

If the screen firmware downgrade is working as you say, I can add it to the description.
Just to get it right, @Crapkin22 you mentioned `Ender-3_ S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER` whereas @m-roberts you mentioned `Ender-3S1_HWv24S1_301_SWV3.0-1.4_C_F401_FDM_LASER`

scubix, 2023-07-25:
> Amazing thanks for this. to add. i read a bit further and saw that my display was TJC (not dacai or Dwin) based on the pics. so i loaded the TJC update (and also put in the DACAI firmware folders [...]

How do you name the firmware file on the SD card and should it be in a STM32F4_UPDATE folder ? I tried all sorts of things, but when turning on the printer, it just sits there with a CREALITY logo on the screen for a while, just like in the opening post, but after a few minutes, it just goes on to boot up and it did not update anything at all. So I have no idea what I am doing wrong. I was able to update the TJC display with the updated firmware as linked in the TJC release tag branch. But the firmware for the printer itself refuses to be flashed

IdanNaor, 2023-08-03:
Worked for me to downgrade to `Ender-3_ S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER`, from version  `Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER`. 
Then I flushed `Ender3S1-F4-UBL-20230522B.bin`.

naedioba, 2023-08-06:
Thank you so much to all of you! 
My display is bak to life. 
I download the the folder Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER  from [here](https://www.crealitycloud.com/software-firmware/firmware?type=2&keyword=&system=0&subType=618ddf638f5f262859ef2091&hardware=0 ), then I used the files "dcboot.bin" first, then "firmware.zlib" with the folder "private".

https://github.com/mriscoc/Ender3V2S1/assets/33942174/84b77f81-de44-4694-9bcb-706640d330fb

mpohoda, 2023-08-07:
I have flashed latest display fw with dcboot.bin and problem with **black screen happens when you replace firmware.zlib** with different file as is in Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER. So now I can only provide downgrade only for private "folder". For dacai_upgrade fw with thumnail preview I took firmware.zlib from Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.1.9_Two_C_F401_FDM_LASER and flashed it with dacai_upgrade private folder.... I'n not happy with this but looks like that works...

Safemode, 2023-08-12:
Just wanted to mention my experience, I've got an Ender 3 S1 with the STM32F401 SoC. Mine came with 3.0.4 on, for some reason wouldn't install the MRiscoC bin out of the box. Ended up updating it to 3.2.0 and then still couldn't get MRiscoC installed, and came across this thread. I was able to successfully downgrade back to 3.0.4 with the firmware linked above. Afterwards I tried to install MRiscoC again with the .bin at the root of the SD Card, and it kept going straight to the Creality main menu. I created a folder on the SD Card called "STM32F4_UPDATE" and put the .bin file in there, and then slightly adjusted the file name with an extra hyphen after Ender to "Ender-3S1-F4-20230522B.bin" (though I don't know that this step did anything, but wanted to include it anyways as I retraced my steps. Seems the more likely fix was putting it inside that folder). Put in the SD Card, and it went through the upgrade process and I'm in business on MRiscoC 2.1.3 20230522 build.

okokel, 2023-08-13:
> I have flashed latest display fw with dcboot.bin and problem with **black screen happens when you replace firmware.zlib** with different file as is in [...]

Had difficulties with the myself, your solution worked for (F4 board, DACAI display).  Have you noticed anything broken with the displays with previews or others?

feelingwalnut, 2023-08-14:
Marlin2.0.8_Ender-3S1_HWv24S1_301_SWV3.2.0_Nine_C_F401_FDM_LASER
does indeed break the ability to install "pro FW"
downgrading only the SCREEN (with the microSD card) to 
Ender-3 S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER
is enough to make "pro FW" functional.

it seems that if you tried to flash pro and got the blank "creality" display, just finishing the process by flashing the screen downgrade is enough to get you functional.
it does not appear that you have to do a full screen and board downgrade or follow a precise order, assuming you allowed enough time for the board to flash in "error" to the blank creality screen.
(i imagine it is possible to improperly flash the board as there is no progress display when trying from most recent official screen FW.)

LeSchmetterling, 2023-08-29:
> Here are the steps I did.
>
> 1. Flashed latest creality screen and motherboard firmware version 3.1.9 the one with dcboot file for the screen since the 3.2 version  didnt work on my printer.. (just to [...]

Could you give more details on this?
Did you just follow the readme for screen and firmware flashing in the download?

regevt, 2023-09-09:
does the solution also work for f103 boards?

BypassTheFedR, 2023-09-22:
Ender 3V2 Neo here. 

Was able to get back to factory by using this firmware

https://www.crealitycloud.com/software-firmware/firmware/ender-series?keyword=&system=0&subType=62e9e331d3227feab178559e&hardware=V4.2.2

I have the TJC Screen but didn't have to mess with that at all. Just had to flash the motherboard with the binary (.bin) file.

baldpope, 2024-03-17:
I believe I've run into this tonight, using an Ender 3 Max Neo - attempted to flash the custom firmware which seemed to work, but lost icons/display on the control panel. tried to go back to the firmware here: https://www.creality.com/pages/download-ender-3-max-neo

but still unable to get the control panel display to flash correctly.

Red369Ender3S1, 2024-03-20:
Ok so I got the Ender-3S1_HWv24S1_301_SWV3.0.4_C_F401_FDM_LASER does the STM32f401RCTx_BootLoader_v1 go on to the display first like the dcboot one or no?

Dennis738, 2024-08-21:
My Ender 3 S1,(F4,DACAI) is getting stuck on the same screen as pic above. I thought it was just bad luck when I bricked my first motherboard (STM32F401RCT6) because I had several versions of self compiled professional firmware on it over a relatively long time period. I tried saving it with precompiled Professional FW and Creality FW, renaming the .bin file each time and putting it in a STM32F4_UPDATE folder…nothing worked. So I bought a replacement Creality STM32F401RET6 motherboard and ran Creality FW for a few days but the options were lacking compared to my other printers that have self compiled firmware. So I compiled Miguel‘s professional firmware and now it’s stuck on the creality screen again lol. The 1st motherboard just quit responding, but this new motherboard has a new failure mode. After each attempted flash it sticks on the creality screen but it also puts an empty DWIN_SET folder in the STM32F4_UPDATE folder with the .bin file. Again I have tried all the pre-compiled FW and professional FW (for the F4 board)with no success. Do you have any suggestions? I’m willing to try anything, it’s already broke and I don’t think I’ll buy another board(because I might be tempted to reflash again). If I can’t fix it, I’ll probably yank off the sprite extruder/cable for one of the other machines. Where can I go to learn about ST-link? I read that somebody had to redo their boot loader after having a similar issue on an ender3S1 F4.

mriscoc, 2024-08-22:
> My Ender 3 S1,(F4,DACAI) is getting stuck on the same screen as pic above...

@Dennis738  Check this: https://github.com/mriscoc/Ender3V2S1/issues/1180#issuecomment-1911528544

Dennis738, 2024-08-22:
Since pre-compiled firmware seems to have the exclusion zone set but currently does not work, should I assume the bootlegger is messed up and working with ST-Link is the only way forward? Then be sure to set the EEPROM exclusion zone if I ever try to compile my own after it’s working again? ST-Link looks doable, I already have the programs to try it.

mriscoc, 2024-08-22:
> it’s working again?

No, if the flags were triggered, that will prevent to flash any other firmware until flags are reset.

Dennis738, 2024-08-22:
Thank you so much. I’ll try to figure out reinstalling the boot loader and comment in the correct topic if things change.

closed 2025-01-05 by github-actions[bot]

thenorhstar, 2025-09-16:
Hello, I'm using the firmware below for Ender 3 v2. It's a great software. First of all, thank you for it. I'd like to ask how I can change the on-screen menus to Turkish (how can I change the language). Can you help me with this?

https://github.com/mriscoc/Ender3V2S1/releases/download/20250106/Experimental-Ender3V2-422-MM-LA-MPC-20250106.bin

vinybazterrica, 2026-10-17:
Hello everyone, I'm having the exact same issue described here.

My setup:

Board: Ender 3 S1 with STM32F103 (V2.4.S1_301).

Display: DACAI screen.

Firmware: Marlin 2.1.4 MRiscoC (UBL-MPC) running fine. I can successfully connect via Pronterface on COM4, and commands like M115, M502, and M500 work perfectly (the printer is fully online and responsive).

The Problem:
After trying to update/flash the official Creality firmware in the past, my DACAI screen bootloader seems to be affected. When I try to flash the dacai_update assets from mriscoc (folder private + firmware.zlib), the screen loads everything correctly, shows update Finished!, and goes blank as expected. However, when I reboot the printer, the screen remains completely black and won't show any interface, even though the mainboard is fully booted and working under mriscoc.

Is there a specific workaround or a patched display firmware package to bypass this post-stock DACAI bootloader restriction? Any help would be greatly appreciated!

mriscoc, 2026-10-17:
This process can help 

1. Get an µSD card of 8GB or less.
2. Format the µSD card MBR, FAT32 and with 4096 as the allocation unit size.
3. Put "dcboot.bin" into the card. 
4. Turn off the printer and insert the µSD card into the card slot at the back of the screen. 
5. Reboot and wait for the update to finish. 
6. After the update, remove the µSD card, delete the "dcboot.bin" and put the "private", "TJC_SET", "DWIN_SET" and "DWIN_SET" folder and firmware.zlib into the Root of µSD card.
7. Turn off the printer and insert the µSD card into the card slot at the back of the screen. 
8. Reboot and wait for the update to finish.
9. After finishing the update, remove the µSD card and delete the files inside.

Use only the files in this zip:
[screen_rescue_firmwware.zip](https://github.com/user-attachments/files/33605957/screen_rescue_firmwware.zip)

mj208287, 2026-10-22:
> This process can help
>
>     1. Get an µSD card of 8GB or less.
>
>     2. Format the µSD card MBR, FAT32 and with 4096 as the allocation unit size.
>
>     3. Put "dcboot.bin" into the card.
>
>     4. Turn [...]

same exact problem here... Can you elaborate more on which printer mainboard flash file  to use with the display files in screen_rescue_firmware.zip. Ive tried just about everything to restore functionality to my ender 3 s1 f4 board. Thank you

mriscoc, 2026-10-22:
>      same exact problem here... Can you elaborate more on which printer mainboard flash file  to use with the display files in screen_rescue_firmware.zip. Ive tried just about everything to restore functionality to my ender 3 s1 f4 board. Thank you

If your printer works, you need only change the display firmware, check the printer with any USB serial console.

kgerakopoulos, 2026-10-26:
same problem here, which firmware with this screen rescue ?? trying 1 week to solve this problem
Ender 3 s1

RandomAnomaly, 2026-11-15:
I was stuck on this for a while, working with claude (yeah yeah, I know) seemed to unblock me, so posting in case it helps out - can't accept any responsibility if it causes you grief, just reporting what I have. I went back to an older version of the mainboard pro firmware. My printer is a STM32F401, with a DACAI display. Symptoms - flashing the pro firmware meant the screen would get stuck at the Creality splash.

What got me unstuck:

1. flash the Ender3S1-F4-UBL-20240126.bin to the mainboard
2. Download the 3.1.9 firmware from creality. Put the dcboot.bin from this package (ONLY) on the microsd, and insert into screen sd slot, boot, allow it to update, then power off
3. Delete dcboot from the microsd, then put firmware.zlib from the 3.1.9 package on the microsd, plus the private/ folder from display assets/dacai_update.zip in the New Year 2026 branch of this repo. Put card in the screen, boot, allow to flash, power off, remove card, boot the printer again.

kgerakopoulos, 2026-11-16:
> requests on the go with GitHub Mobile for iOS
> and Android
> Download it today!

Thanks for all the tips.... from a black screen for a long time now i am
with the stuck Creality logo as you....well it is a progress.

I repeated for the second time the process but i am stuck with the logo

RandomAnomaly, 2026-11-16:
What is your chip version? Is it a f401 board?