mriscoc/Ender3V2S1 #968: [BUG] Multiple_Probing, M48 ProbeTest, High_Speed_Mode

classicrocker883, 2023-06-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I would like to have multiple_probing to do a double probe (two times touch), but this doesn't seem possible since the update. A value of 0 is as expected - does not add additional probes. but a value of 1 cannot be done, and even if it is changed in the firmware by silencing the error, it still acts as though it is a value of 0. so the next value 2 adds 2 extra probes, making it a total of 3. the options are either 0 = 1 probe, or 2 = 3 probe. or more, and so on. Is there a way to make it so 1 = 2 probe? 

`EXTRA_PROBING` doesn't seem to do anything if it is enabled or not (by commenting out the error in the code)

now the issue with M48, there should be a way so there is `NothingToDo`, because if you were to click M48 Probe Test more than once, it will continue over. is there a way so the screen can freeze like with the Homing Popup?

and Highspeed mode: the issue is when it is disabled, it doesnt raise after a probe. so what happens is the probe keeps hitting the bed and causing a stop error. i could make a video of this happening for more detail.

### Version of Professional Firmware

current May update

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-06-23:
This a requirement:
https://github.com/mriscoc/Ender3V2S1/blob/73c11535a7c0de0044cfed0f564154a079be5782/Marlin/src/inc/SanityCheck.h#L1433-L1439

mriscoc, 2023-06-23:
Please open a new issue elaborating this:

> and Highspeed mode: the issue is when it is disabled, it doesnt raise after a probe. so what happens is the probe keeps hitting the bed and causing a stop error. i could make a video of this happening for more detail.

closed 2023-06-23 by mriscoc

classicrocker883, 2023-06-23:
it doesn't appear that EXTRA_PROBING has an effect, even I commented out the error from showing and disable in the file. 

what I mean when HS mode is disabled I mean from the LCD menu. and this doesnt allow the probe to raise for multiple probes.

also, when M48 is called the probe starts at any point on the bed. it should be at the middle, so "G28Z" should be added to the `ProbeTest()` function. also a pause flag so the user can't use the lcd while it's doing M48 probing.

mriscoc, 2023-06-24:
> ... even I commented out the error ...

You should never comment out the SanityCheck.h requirements, they were written there for a reason.

> what I mean when HS mode is disabled I mean from the LCD menu. and this doesnt allow the probe to raise for multiple probes.

You can control how much Z is raised by changing the Z Clearance:
https://github.com/mriscoc/Ender3V2S1/blob/73c11535a7c0de0044cfed0f564154a079be5782/Marlin/Configuration.h#L1629-L1632

> also, when M48 is called the probe starts at any point on the bed. it should be at the middle,...

No, [M48](https://marlinfw.org/docs/gcode/M048.html) can be run anywhere over the bed similar to G30, you can use the X and Y parameters to move the probe to specific point.