mriscoc/Ender3V2S1 #969: Bed temp drops to 0

KBalage, 2023-06-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I've tried to print ASA with bed temp 100°C, Hotend 260, had an E3S1+ with firmware Ender3S1-F4-UBL-20230522B.bin, PrusaSlicer 2.6.
My starting G code sets the heat for bed to target and hotend first to 160 only, then after levelling steps (only load mesh) it sets the hotend temp to 1st layer temp. 

The problem is that the bed temp never reach 100, and a new target - 0 - is set by some reason. By the time the hotend reach the set temp and printing starts, the bed has a 0 target, slowly cools down and the print is messed up.

### Version of Professional Firmware

Ender3S1-F4-UBL-20230522B.bin

### Printer model

E3S1+ F4

### Electronics

F4

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

### Additional information & file uploads

[image]

mriscoc, 2023-06-23:
Compare your start script with: https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts

KBalage, 2023-06-23:
> Compare your start script with: https://github.com/mriscoc/Ender3V2S1/wiki/Slicer-G-code-Scripts
I did, and I even checked the generated G code for M140 commands - there weren't any - apart from the first one which set the bed temp to 100. 
Interesting - if I start the printer, pre-heat the bed (close to 100) it works fine. If I start with cold bed, it fails just like I described. 
There's a timeout somewhere...

mriscoc, 2023-06-23:
Yes, there is a timeout if the nozzle is heated and the extruder doesn't move, because that can causes that filament residues to crystalize and clog the nozzle.

KBalage, 2023-06-24:
Thank you, it explains now what's going on here.

PrusaSlicer by default heat the extruder to 170°C and then wait till bed leveling temp. After the leveling done it continues with heat up the extruder to first layer temp.
In case we have a slow performance in bed heating (like E3S1+ with 100°C target) it leads to the timeout on both extruder and temp, but extruder temp later set to first layer temp while the bed is turned off. 

I just have to reorder the start G code a bit, wait till bed temp, wait till extruder leveling temp and then continue. 

M862.3 P "[printer_model]" ; printer model check
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S170 ; set extruder temp for bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M204 T[machine_max_acceleration_travel] ; restore travel acceleration
M104 S[first_layer_temperature] ; set extruder temp
G92 E0
G1 Y-2 X179 F2400
G1 Z3 F720
M109 S[first_layer_temperature] ; wait for extruder temp

; intro line
G1 X170 F1000
G1 Z0.2 F720
G1 X110 E8 F900
G1 X40 E10 F700
G92 E0

M221 S95 ; set flow

closed 2023-06-25 by KBalage