mriscoc/Ender3V2S1 #975: [BUG] After changing X bed size and Y bed size to 400, probing with crtouch doesn't compensate for settings

melsbacksfriend, 2023-06-27:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I installed the ender extender 400xl and changed x bed size, y bed size, z max position, x max position and y max position to compensate and the bed levelling does not compensate for that.  I made sure to disable HS mode.  

### Expected behavior

I expect the probe to reach the whole bed

### Actual behavior

It only goes up to the stock bed size

### Steps to Reproduce

1. Install ender extender and crtouch
2. Reconfigure settings as I did
3. auto build mesh

### Version of Professional Firmware

2.1.3 MRiscoC 20230522

### Printer model

Creality Ender 3V2

### Electronics

4.2.2 board

### Add-ons

ender extender, crtouch, filament runout sensor

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-06-28:
Check the mesh insets https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

melsbacksfriend, 2023-07-11:
Thanks!  That helped a lot!

closed 2023-07-11 by melsbacksfriend