mriscoc/Ender3V2S1 #979: [DOC] Cannot update FW on MacOS

BogdanTheGeek, 2023-07-01:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

No way to update the firmware from a Mac.

### Bug Timeline

New macOS?

### Expected behavior

Expect to just format the SD card and copy the file.

### Actual behavior

Blank Display

### Version of Professional Firmware

2.1.3

### Printer model

3V2

### Electronics

4.2.2

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

new versions of macOS seem to break the device. I think this is due to macOS adding multiple files when a simple copy is performed. I have managed to update my firmware by deleting all of the hidden dot files from the terminal then ejecting the sd card. I think it would be useful to document this behaviour in order to help others.

gothcha, 2023-07-01:
There is no reason why the update will not work with MacOS, I use it and update perfectly. It may be due to the format of the SD card, there are many discussions on this available to ensure that the correct formatting is used for the 3V2

BogdanTheGeek, 2023-07-01:
I have formatted the sd card with disk utility, sd card formatter and on
linux. The format is not the issue, it turned out to be my M1 mac, running
the latest macOS, creating a bunch of random hidden files like .Spotlight
and ._fwirmare.bin
I didn't post this looking for help, but to help others. Feel free to close
it if you feel like the documentstion is sufficient and It was user error
on my part.

mriscoc, 2023-07-01:
Hi, thanks for your observations.
The "Issues" section is usually used for reporting bugs, perhaps the "Discussion" section might be a better place for your findings.

kpbicka, 2023-08-25:
>  ._fwirmare.bin *>

Use USBClean.app for Mac or another you like

mriscoc, 2023-09-06:
Not a firmware issue

closed 2023-09-06 by mriscoc

mriscoc, 2025-05-21:
Discussion related to this issue: https://github.com/mriscoc/Ender3V2S1/discussions/1451