mriscoc/Ender3V2S1 #980: [BUG] Firmware cannot initialize with Creality Sprite Extruder Pro

mobigroup, 2023-07-02:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Firmware Ender3V2-427-BLTUBL-MPC-20230312.bin does not start; BL touch looks initialized but the loading freezing on the screen before the main menu. I don't need 300 degree limit for now so there is no reason to change the firmware.

### Version of Professional Firmware

BLTUBL-MPC-20230312

### Printer model

Creality Ender 3V2

### Electronics

4.2.7

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-07-03:
Check your connections, this firmware works with the Creality Sprite Extruder Pro.

closed 2023-07-03 by mriscoc

mobigroup, 2023-07-06:
I’ve already compiled custom firmware for the extruder and the main board and it works (tested up to 250 degrees as my filaments allow). I think it would be nice to make the initialization possible always and show informative errors after that.