mriscoc/Ender3V2S1 #982: [BUG] (Cannot compile FW)

drradkov, 2023-07-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

When try to compile firmware, error appears:
   77 |   #error "HAS_GCODE_PREVIEW requires PROUI_EX."
   80 |   #error "HAS_TOOLBAR requires PROUI_EX."
      |    ^~~~~
Error: Failed to parse Marlin features. See previous error messages.

### Bug Timeline

New

### Expected behavior

.

### Actual behavior

.

### Steps to Reproduce

.

### Version of Professional Firmware

Latest

### Printer model

Creality Ender 3V2

### Electronics

Board 4.2.7

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

OctoPrint

mriscoc, 2023-07-05:
You need to follow that requirements
```cpp
77 | #error "HAS_GCODE_PREVIEW requires PROUI_EX."
80 | #error "HAS_TOOLBAR requires PROUI_EX."
```

closed 2023-07-05 by mriscoc

drradkov, 2023-07-05:
Could you please sugdest, how to fix it?

mriscoc, 2023-07-05:
It is not an bug, it is a requirement. If you want to use the toolbar, for example, you need to enable PROUI_EX.

drradkov, 2023-07-05:
Buh how? i tied to search the repo. Nothing found. 
Thanks.

mriscoc, 2023-07-06:
It is enabled by default in the provided configuration files.

drradkov, 2023-07-06:
Thank you. Somehow i fownloaded and old version os configurator. After fowloading the lates py script, evrything is compuling without issues.

Vodlex1, 2023-07-31:
Is there a way that I could disable the toolbar and gcode previewer? And would doing that cause other problems?

mriscoc, 2023-08-01:
> Is there a way that I could disable the toolbar and gcode previewer? And would doing that cause other problems?

Yes, look for:

```cpp
  #define HAS_GCODE_PREVIEW 1
  #define HAS_TOOLBAR 1
```

In the Configuration.h file.