mriscoc/Ender3V2S1 #990: CURA 5.4 no Jpeg Preview

jxer, 2023-07-14:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Upgraded to 5.4.0, script and post processing tasks all look correct but the ender is showing the qr code

### Version of Professional Firmware

2 revisions back from latest

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

mriscoc, 2023-07-15:
Update your firmware, Seems that Cura 5.4 is working ok:   

[image]

classicrocker883, 2023-07-19:
I'm also actually not getting a preview image either using Prusa Slicer. it just shows blank. it also takes a second longer to load, the other files and gcode sliced in Cura have no issue.

Could it be something with the .py file? or the parameters entered in the slicer itself?

Antufatiko, 2023-07-25:
Did you fix the problem? @jxer i have the same and don't know what to do

classicrocker883, 2023-07-26:
is it possible that it wont work because the size value you are using is unsupported? so maybe it needs to be changed to another value.

Antufatiko, 2023-07-27:
> is it possible that it wont work because the size value you are using is unsupported? so maybe it needs to be changed to another value.

I did everything the post said but in Cura 5.4 doesn't appear the "Create JPEG thumbnail".
I did it in Cura 5.2.1 and it works.

mriscoc, 2023-07-27:
> I did everything the post said but in Cura 5.4 doesn't appear the "Create JPEG thumbnail".

[image]
[image]

closed 2023-07-27 by mriscoc

classicrocker883, 2023-07-28:
when you installed the new Cura, did you make sure to paste the slicer script *for Cura* in the correct folder?

that's how to get the create jpeg thumbnail to appear