mriscoc/Ender3V2S1 #992: [BUG] Retraction length for Firmware Retraction is limited to 10mm

bronozoj, 2023-07-17:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

It's not really that much of a bug per se but the firmware retraction settings have a limit on the retraction length. Is this by design? I could bypass this artificial limit by issuing an M207 Sxx where it will reflect on the menu but trying to modify this setting using the menu will snap it right back to 10mm.

### Version of Professional Firmware

 Ender3V2-422-BLTUBL-MPC-20230522

### Printer model

Creality Ender 3V2 Neo

### Electronics

4.2.2 GD32

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-07-17:
Firmware retraction settings is limited to 10 mm when it is edited by menu
```cpp
void setRetractLength() { setPFloatOnClick( 0, 10, UNITFDIGITS); }
```

bronozoj, 2023-07-18:
Is it going to be the default limit or is it something that can be considered to be changed?

mriscoc, 2023-07-18:
It is the default limit and can be easily changed by editing the line posted before and compiling a custom version. Why do you need more than 10 mm for a firmware retract?

bronozoj, 2023-07-18:
I have been attempting to use TPU fillament with the bowden extruder and reducing stringing by increasing retraction distance. Since this isn't really a bug, I will be closing this issue.

closed 2023-07-18 by bronozoj