mriscoc/Ender3V2S1 #994: [BUG] Unable to compile source when specifying CR10_StockDisplay defined

josurr, 2023-07-19:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

The firmware is being compiled with "CR10_StockDisplay" being defined in order to use the BTT TFT70 3.0 LCD.

My Ender 3v2 is using an SKR MiniE3 v3 mainboard.

The following compile terminal errors occur:

```
In file included from Marlin\src\lcd\dogm\status_screen_DOGM.cpp:34:
Marlin\src\lcd\dogm\../marlinui.h: In static member function 'static void MarlinUI::draw_status_message(bool)':
Marlin\src\lcd\dogm\../marlinui.h:372:17: error: 'MarlinUI::status_message' has incomplete type
  372 |     static char status_message[];
      |                 ^~~~~~~~~~~~~~
Marlin\src\lcd\dogm\status_screen_DOGM.cpp:918:18: error: invalid use of array with unspecified bounds
  918 |           slen = status_message.glyphs();
      |                  ^~~~~~~~~~~~~~
Compiling .pio\build\STM32G0B1RE_btt\src\src\lcd\menu\menu_media.cpp.o
Compiling .pio\build\STM32G0B1RE_btt\src\src\lcd\menu\menu_mixer.cpp.o
Compiling .pio\build\STM32G0B1RE_btt\src\src\lcd\menu\menu_mmu2.cpp.o
*** [.pio\build\STM32G0B1RE_btt\src\src\lcd\dogm\status_screen_DOGM.cpp.o] Error 1
```

### Expected behavior

Compile successfully.

### Version of Professional Firmware

02010300

### Printer model

Ender 3v2

### Electronics

SKR MiiniE3v3

### Add-ons

BTT TFT70 v3.0

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

None

mriscoc, 2023-07-19:
https://github.com/mriscoc/Ender3V2S1/wiki#requirements

classicrocker883, 2023-07-20:
try deleting  
framework-arduinoststm32@src-06b9d2668dcd05a92820ac50e10b17c7  
framework-arduinoststm32@0.0.0  
or whatever the folder is called from `C:\Users\$USER\.platformio\packages`

here is a copy of my configs for this. *use for reference only**
[BTT-SKR_Mini_E3_V3.zip](https://github.com/mriscoc/Ender3V2S1/files/12101909/BTT-SKR_Mini_E3_V3.zip)

look through it see if any options help you out

closed 2023-09-11 by mriscoc