mriscoc/Ender3V2S1 #999: [BUG] Extruder stops after some time with LA firmware

fraengers, 2023-07-23:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

I'm trying to use Linear Advance but the extruder stops (stepper off) after some time (usually during the first few layers). 
I'm using Ender3V2-422-MM-LA-MPC-20230524.bin printing from Octoprint and SD-card (no difference).
When I use Ender3V2-422-MM-MPC-20230522B.bin there is no problem, so the gcode itself works. 

### Version of Professional Firmware

Marlin 2.1.3 MRiscoC Ender3V2-422-MM-LA-MPC, based on bugfix-2.1.x (May 24 2023 16:09:04)

### Printer model

Ender 3V2

### Electronics

4.2.2

### Bed Leveling

MBL Manual Bed Leveling

### Your Slicer

Prusa Slicer

### Host Software

OctoPrint

earthwormjimwow, 2023-07-26:
Same thing happens to me.  I have a 4.2.2 board with TMC2208 drivers.  This is related to stealthchop being forced on and most likely having TMC2208 drivers in your board; people with TMC2209 drivers seem to not have problems.  Since these drivers are in standalone mode, we can't disable stealthchop and enable spread cycle.  

Because of a hardware bug or limitation in TMC2208 drivers related to over-current detection and stealthchop, this leads to false detection of over-current events when the minimum time between steps is too short.  Linear advance is going to require very fast but small movements, which leads to false over-current protection, shutting down the stepper motor driver until reset.

It's doubtful this can be fixed in firmware, and is not a firmware bug at all in Marlin firmware.  You need to get a 3rd party board, or do a UART mod so you can configure your stepper driver and disable stealthchop, and change it to spread cycle.

This hardware bug/limitation is alluded to in the Apogee tool head firmware documentation.
https://www.orbiterprojects.com/apogee-ender-3-v2/

mriscoc, 2023-07-29:
@earthwormjimwow do you have a link to some recent posted issue in the Marlin Github with reference to this problem. I remember old issues with LA deprecated implementation but I can't find information about this issue in with new LA implementations.

mriscoc, 2023-07-29:
@fraengers remember to disable in the slicer any acceleration/jerk control or coasting modifiers.

earthwormjimwow, 2023-07-29:
@mriscoc does your latest release include this Marlin improvement?
https://github.com/MarlinFirmware/Marlin/pull/25696

If so, then I still have the same freezing issue with the extruder stepper.  My printer does not completely crash or halt though, just the extruder stepper stops until I power cycle the printer, due to over current protection kicking in.

I see posts saying the issue is "fixed" but it never actually results in a usable printer that can print for more than an hour before crapping out, and I never see anyone actually showing proof of a working printer.

I really doubt this can be fixed in firmware alone, since linear advance requires fast changes in movement, otherwise it's not linear advance, and TMC2208 drivers have false tripping issues with fast steps in stealthchop mode...

mriscoc, 2023-07-29:
> does your latest release include this Marlin improvement?
> [MarlinFirmware/Marlin#25696](https://github.com/MarlinFirmware/Marlin/pull/25696)

Yes, it is included, all PR in the bugfix until May'23 are included.

fraengers, 2023-07-30:
@earthwormjimwow that seems plausible. ''ll look into the UART mod when I find the time.

fraengers, 2023-07-30:
@mriscoc it's hard to tell, which setting are relevant (I'm using Prusa Slicer), but I'll try. 
Do you have a working test gcode for LA firmware, by any chance?

mriscoc, 2023-07-30:
In the Telegram test group, some users are using LA builds on their 4.2.2 board with TMC2208 drivers without issue.

ZackySteel, 2023-08-05:
> In the Telegram test group, some users are using LA builds on their 4.2.2 board with TMC2208 drivers without issue. But they are using the latest experimental version which has updated fixes.

I've been searching for why this same issue might be occurring for me, came across this post, and finally realized it was only after I enabled linear advance on my Ender 3 V2 Neo that the long prints I ran after testing and optimizing settings began running into this issue! Being the internet-Neanderthal that I am, I'm completely clueless as to the workings of Telegram and can't seem to find any information on the Telegram group aside from the occasional mention on Reddit threads relating to the firmware. I'm currently in a holding pattern printer-wise because I feel like I've hit a bit of a ceiling quality-wise without linear advance, input shaping, etc. and was waiting to see if there was any solution to this issue aside from disabling these features for the time being.

If anyone can shed some light it'd be greatly appreciated! 

Thanks!

earthwormjimwow, 2023-08-05:
Maybe I am wrong that this can't eventually be solved with firmware.  I gave up trying to get it to work and switched to Klipper running on my Pi 3.  Pressure Advance works fine via Klipper, so maybe there's still hope for Linear Advance eventually with these standalone TMC2208 drivers?  

Not sure why Klipper's implementation isn't causing stepper driver over current protection to kick in.  Maybe it needs less jerky movements from the extruder compared with Marlin's implementation?  Maybe more processing power, so Klipper can look further ahead and take better action accordingly.

Either way, definitely sticking with Klipper.  So much easier to tweak settings.  Not only that but I'm able to print at better quality than I ever achieved before with about a 50-60% print time reduction.  Now using 6000 mm^2/s accelerations and 200 mm/s max linear speeds.  

Previously on Marlin, the fastest accelerations I could manage even with input shaping were 3500 mm/s before stepper skipping, and no faster than 150mm/s linear speeds before stepper skipping.    Actually couldn't really go past 100 mm/s print speeds, since quality nose dived any faster since I couldn't use linear advance.

I haven't even done my planned stepper UART mod, so my stepper drivers still have stealthchop, yet no issues with Pressure Advance.   I eventually plan to do the UART mod though, since I want more motor torque, since i get skipping if I try to  print faster.  The LCD display header is now freed up to be used for UART connections to the stepper drivers.

ZackySteel, 2023-08-06:
> Maybe I am wrong that this can't eventually be solved with firmware. I gave up trying to get it to work on Marlin and switched to Klipper running on my Pi 3. Pressure Advance works fine via Klipper, so maybe there's still hope for Linear Advance eventually with these standalone TMC2208 drivers?
> 
> 

I had figured that was the case for a while, but then pretty much everything I've read within the past year or two indicates that an update actually did fix the bug causing it, so I'm definitely confused! I see he mentioned the Telegram group reporting success with some more recent builds and I've seen others on forums who confirm that yes, it had been an issue for a long time, but then also say that yes, an update fixed that issue and linear advance now works properly. I finally decided to dive back into the wide-world of firmware despite already loving the professional firmware I had managed to install mainly because everything I've been reading "confirms" that the issue has been dealt with. Of course despite everything I read, I'm still running into the exact problem one would expect were the issue still present, so I'm definitely feeling thoroughly clueless at this point!

If anyone knows of any newer builds that they've been having any luck with, I'd love to know!

Thanks!

earthwormjimwow, 2023-08-06:
> I had figured that was the case for a while, but then pretty much everything I've read within the past year or two indicates that an update actually did fix the bug causing it, so I'm definitely [...]

Perhaps for some people, the latest Marlin firmware resolves the issue, because of a combination of their extruder, bowden vs. direct drive setup, print speeds, LA settings required, and build date or even sample variation of their TMC2208 drivers.  

It's entirely possible we are dealing with a marginality problem related to a bug/defect/sensitivity in the TMC2208 drivers.   For some users the problem is now gone, for others, the false over current tripping can still happen.  It could even be a probability related issue at this point.  Perhaps every step with an extremely short time window has X% chance of tripping over current protection.  On some samples of TMC2208 drivers, this probability could be 0.0000001%, on others its could be 0.000001%.  So if the people who have this issue resolved printed for thousands of hours, maybe they would see an over current trip event at least one time.  For us though, it only takes an hour or so of printing.

We like to think that mass produced ICs are all interchangeable and identical, but there is always sample to sample variation, that roots its ugly head in bugs or edge cases.  I think this Linear Advance issue is one of those cases.

On the power supplies I work on, we've had issues where random noise spikes can cause false over current tripping of the power supply's controller IC.  Swap out the IC for a different sample, even of the same batch, and the same spikes don't cause a false over current trip.

ZackySteel, 2023-08-06:
> Perhaps for some people, the latest Marlin firmware resolves the issue, because of a combination of their extruder, bowden vs. direct drive setup, print speeds, LA settings required, and build date or even sample variation of their TMC2208 drivers.
> 
>
 Yeah, I'd love to get more specific info from those who have it working, 'cause it definitely doesn't seem to be a topic where any kind of definitive conclusion has been reached. I'm cautiously optimistic hearing that some success is being found with the newer experimental build, but until I manage to find that version, it looks like my printer is going to remain in hibernation! If anyone knows more about the experimental build front, I'd love to know more! 

Thanks again Mriscoc for all your work on this firmware, the visualization for mesh generation and bed-tramming alone has improved my quality of printing life more than I can say! I so much appreciate your efforts!

ZackySteel, 2023-08-23:
> In the Telegram test group, some users are using LA builds on their 4.2.2 board with TMC2208 drivers without issue. But they are using the latest experimental version which has updated fixes.

Any updates from the test group? I keep hearing that some 4.2.2 TMC2208 printers work fine while others suffer from the same issues they did before the update. I'm very interested to hear more as to whether or not there are similar issues with some working but others not despite seemingly almost identical specs!

dfinberg, 2023-08-23:
I've moved to klipper, but I had the same extruder lockup issue on a 422 T8 with the may release baseline (i had compiled my own version ).  I have not had any issues on klipper. Certain LA settings seemed to make it much worse, but it was pretty consistent across the spectrum.

closed 2023-11-03 by github-actions[bot]

mriscoc, 2024-02-20:
This discussion could be related to this issue: https://github.com/mriscoc/Ender3V2S1/discussions/1259