mriscoc/Ender3V2S1 #1003: [BUG] Ender 3 S1 not using bed mesh

bauch1234, 2023-07-26:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Ender 3 S1 F4 not using the bed mesh during print 
This is my start gcode
; Ender 3 Custom Start G-code
G28 ; Home all axes
M420 s1 z10 ; Use saved bed mesh
G92 E0 ; Reset Extruder

### Expected behavior

i used command m420 it should load and use the saved mesh

### Actual behavior

it doesnt use the mesh at all

### Steps to Reproduce

make mesh and add m420 command to start gcode

ignores mesh and does what it wants anyway 

### Version of Professional Firmware

precompiled Ender3S1-F4-UBL-20230522B.

### Printer model

s1

### Electronics

f4

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

classicrocker883, 2023-07-27:
try 
>M420 S1 L0 Z10 ; Use saved bed mesh

youre using Lower case, ALSO it may need to Load the correct slot. `L0` would be Load slot 0, this can be which ever slot you choose.

mriscoc, 2023-07-27:
> M420 is for UBL,
> G29 A is for ABL/ bilinear

M420 S1 is for ABL/UBL
G29 A is only for UBL 

both need to be the last leveling command

mriscoc, 2023-07-27:
> M420 s1 z10 ; Use saved bed mesh

Those parameters  must be in capital letters

bauch1234, 2023-07-27:
okay i updated the command to M420 S0 Z1,  and made a new mesh right before printing and still doesn't use the mesh when printing

bauch1234, 2023-07-27:
M420 S1 Z10 did false alarm

mriscoc, 2023-07-28:
> okay i updated the command to M420 S0 Z10, and made a new mesh right before printing and still doesn't use the mesh when printing

S0 -> disable
S1 -> enable

bauch1234, 2023-07-28:
i thought that was the save slot yea s1 worked thats whjat, how much would i have to donate to get your help compiling firmware with the settings as listed on configurator Printer: ender 3 s1, board: 301f4, leveling BLT with UBL, screen TJC, thermistor T1, features LA, MPC when i use these settings i get an error in VSC in the stm32f401rc_creality 256k environment, it will work in the stm32f401re_creality 512k environment but it wont work on my printer

classicrocker883, 2023-07-28:
> i thought that was the save slot yea s1 worked

as I wrote earlier
> `L0` would be Load slot 0, this can be which ever slot you choose.

remember L for Load, I suppose S for Start?

for your printer, you have stm32f401rc_creality 256k? it will work for 512k because it has more memory storage. 256k is not enough for LA, MPC ect. I suppose it can but you will have to remove some options to save memory to allow it to fit.

mriscoc, 2023-07-29:
> i thought that was the save slot yea s1 worked thats whjat, how much would i have to donate to get your help compiling firmware with the settings as listed on configurator

Here is how to request a custom build: https://github.com/mriscoc/Special_Configurations#how-to-request-a-custom-build

closed 2023-07-29 by mriscoc