mriscoc/Ender3V2S1 #1006: [BUG] BLtouch Mesh negative Z on May Release

dymester, 2023-07-30:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Hi, I have recently updated from 20230312 to 20230522 and found myself having difficulties leveling and tramming my bed.
At the start I though I bent something but no. Mesh building (9x9 and 5x5) always came up as almost perfect level around the center (sorry no images, but I can reinstall 20230522  to get some if needed), between 0.01 and -0.02 but 3 probe points away from the center (on a 9x9) would go to -0.10 and in the farthest points, -0.20. To make double sure I probed my glass bed and even the bare metal, always "deep" negatives on the same probing points

I downgraded back to 20230312 without touching the knobs or even leting the bed cool down, meshed, and the problem was no more. with a mesh running around -0.01 and 0.07, normal values for my printer.

### Bug Timeline

New issue, since installing 20230522 

### Expected behavior

Mesh building results to be consistent between versions

### Actual behavior

Mesh on 20230522 with big -Z bias on outer edge of bed

### Steps to Reproduce

Starting on 20230522 
1. Heat bed to 60
2. Trammed (tramming results would always tend to be negative)
3. Built mesh on glass, negative values on outer ring
4. Built mesh on bare metal, negative values on outer ring
5. Installed 20230312
6. Heat bed to 60
7. Set up my probe offsets
8. Run tramming wizard, did not touch adjust knobs. (This time values are positive)
9. Built mesh on bare metal, normal values in mostly on positive values
10. Built mesh on glass, also normal values

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC-20230522B

### Printer model

Creality Ender 3V2

### Electronics

Board 4.2.2

### Add-ons

Original BL Touch

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Additional information & file uploads

No additional changes where done to the printer

classicrocker883, 2023-08-03:
How about your Z-Offset? is that different between build versions?

mriscoc, 2023-08-03:
Maybe related to https://github.com/mriscoc/Ender3V2S1/discussions/976

mriscoc, 2023-08-15:
Possible solution: https://github.com/mriscoc/Ender3V2S1/discussions/1021

dymester, 2023-08-21:
Hey, sorry for the radio silence, this indeed fixes the issue. Thank you very much

> Possible solution: #1021

closed 2023-08-21 by dymester