mriscoc/Ender3V2S1 #1008: mesh bed leveling fails at probe point 48

xx1180, 2023-07-31:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

after upgrading to the latest version i have come across a problem and i don't know how to solve it
all was working fine before the update. i am running ender 3 s1, s1pro and s1plus and this problem is on all 3. when running the auto bed mesh the mesh fails at probe 49 on a 7x7 grid the probe retracts and homes then tries to restart the mesh probing. this problem has not happen if i do a 6x6 grid mesh. can you please help

### Bug Timeline

new bug after updating to the latest version

### Expected behavior

7x7 grid mesh probing

### Actual behavior

fails at probing number 48 of 49 7x7 grid

### Version of Professional Firmware

version 2.1.3 F4 T13

### Printer model

s1 F4 board

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Cura

### Host Software

Cura

classicrocker883, 2023-08-03:
i think it has to do more with your Mesh Inset. its probably failing because its a point it cannot reach. what is your Probe Offsets? and what are your Mesh Insets?

mriscoc, 2023-08-03:
Maybe you have a CRTouch and forgot to disable HS mode.

xx1180, 2023-08-03:
yes that was what was wrong turned off HS and it worked first time thanks

classicrocker883, 2023-08-03:
> yes that was what was wrong turned off HS and it worked first time thanks

I have a CRtouch and haven't experienced this with HS enabled. can that be explained more what HS has to do with CR touch?

mriscoc, 2023-08-04:
> I have a CRtouch and haven't experienced this with HS enabled. can that be explained more what HS has to do with CR touch?

https://github.com/mriscoc/Ender3V2S1/wiki/3D-BLTouch#notes-about-crtouch

closed 2023-08-10 by mriscoc