mriscoc/Ender3V2S1 #1011: [BUG] Mesh leveling and Tramming offset

MartinBaadsgaard, 2023-08-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Bed leveling and tramming appear to not position the probe within/at edges and corners of a square that matches the bed.
With bed leveling maybe this is accounted for, but for bed tramming it gives an incorrect reading.

### Bug Timeline

Has been observed at least since Dec. 22

### Expected behavior

All probings for Tramming or all edge probings for Mesh Leveling are same distance from edge of bed regardless of whether it is at max or min of X or Y.

### Actual behavior

Probing at max coordinates are offset further inwards from bed edge than at min coordinates.
(Appears as if the coordinates are for the nozzle's min and max values, disregarding the probe offset)

### Steps to Reproduce

**Pre-steps:**
Ensure physical bed dimensions and movement min and max are correct in firmware, to an extent where min/max of X/Y are somewhat equal distance from each edge of bed.

Ensure probe offset settings matches real world by measure or test-print.

**Steps:**
Activate either Tramming Wizard or Auto Build Mesh.

### Version of Professional Firmware

20230805

### Printer model

E3 S1 F1 

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

mriscoc, 2023-08-06:
Check https://github.com/mriscoc/Ender3V2S1/wiki/Calibration-Guides

classicrocker883, 2023-08-06:
adjust your Probe Offsets. you want the nozzle and probe tip when deployed to line up, so the probe 'becomes' the nozzle. so as long as your nozzle is on the bed, so will the probe.

https://youtube.com/shorts/FKvPU2nwdts?feature=share

closed 2023-09-11 by mriscoc