mriscoc/Ender3V2S1 #1012: Home screen invisible bottom functions

Juli337-afk, 2023-08-05:
### Did you test with a precompiled firmware?

Yes, and the problem still exists.

### Bug Description

Before updating to the latest release, on the homescreen were about 5 options on the bottom (Home, disable stepper, create mesh,etc) Now they are invisible, I can't see them but if I turn the knob all the way to the right and press it, it will disable the stepper motors.

### Bug Timeline

Brand new, happened today with the new update (AUG 8)

### Expected behavior

Homescreen as usual

### Actual behavior

Bottom buttons invisible

### Steps to Reproduce

Just update to the latest version of the firmware (I also updated the DWIN_SET on the screen)

### Version of Professional Firmware

Ender3V2-422-BLTUBL-MPC

### Printer model

Ender 3 V2

### Electronics

4.2.2

### Add-ons

CR Touch

### Bed Leveling

UBL Bilinear mesh

### Your Slicer

Cura

### Host Software

SD Card (headless)

classicrocker883, 2023-08-06:
did you compile your own?  
https://github.com/mriscoc/Ender3V2S1/issues/249

it disappears when you change the author, that should be left alone

Juli337-afk, 2023-08-06:
> did you compile your own? #249
> 
> it disappears when you change the author, that should be left alone

Nope, I just downloaded the new release and installed it. The LA version also wont show those buttons.

Crazy-Charles, 2023-08-06:
> did you compile your own? #249
> 
> it disappears when you change the author, that should be left alone

Not anymore, authors name is embedded in the proui libraries.

Juli337-afk, 2023-08-06:
This is how the menu looks like for me 
[image]

mriscoc, 2023-08-06:
Some people are reporting the same issue, but I if load the file: https://github.com/mriscoc/Ender3V2S1/releases/download/20230805/Ender3V2-422-BLTUBL-MPC-20230805.bin
in my printer all seems correct.
What SoC your printer has? STM32 or GD32?

Crazy-Charles, 2023-08-06:
Just tried the precompiled Ender3V2-427-BLTUBL-MPC-20230805.bin file and it has the same problem, no toolbar or proUIEx.init display.

mriscoc, 2023-08-06:
Please post the results of **C115**

Juli337-afk, 2023-08-06:
My SoC is yhe STM32

Juli337-afk, 2023-08-06:
> Please post the results of **C115**

What's that

Crazy-Charles, 2023-08-06:
>>> C115
SENDING:C115
Marlin ProUI Version: 2.1.3 MRiscoC
FFICIAL BUILD: 20230805 - 06:59
Author: Miguel Risco-Castillo (MRiscoC)
ok P15 B15

RejZoRSheep, 2023-08-06:
I'm using latest BLT 422 version and the toolbar is invisible indeed. It seems to be there as turning Ender 3 v2 knob to select icons seems to go further, you just don't see it.

renepetro, 2023-08-06:
Seems like to be some inconsistencies in config files , which one is correct ?

#define STRING_CONFIG_H_AUTHOR "Miguel Risco-Castillo (MRiscoC)" // Who made the changes.

#define STRING_CONFIG_H_AUTHOR "Miguel A. Risco-Castillo (MRiscoC)" // Who made the changes.

mriscoc, 2023-08-07:
**NOTE**: Due to a bug, where some people report that the info box and toolbar are not showing, binary downloads have been removed. The error seems to be board dependent, some printers with the same firmware file work and some don't. Until this can be fixed, the binaries will not be released.

mriscoc, 2023-08-15:
Please download the more recent binaries, the issue was solved.

closed 2023-08-15 by mriscoc